JMRI/JMRI #13381: Z21XPressNet uses Z21LocoNet for POM

higaski, 2024-08-26:
Picking the option "Programming On Main" through a radio button in the lower right corner while having "Z21XPressNet" as programming mode results in JMRI sending LocoNet messages.

[image]

[image]

The expected behavior would be that JMRI sends a [LAN_X_CV_POM_READ_BYTE](https://www.z21.eu/media/Kwc_Basic_DownloadTag_Component/root-en-main_47-1652-959-downloadTag-download/default/d559b9cf/1628743384/z21-lan-protokoll-en.pdf) command.

danielb987, 2024-08-26:
@pabender 
Is this something for you to take a look at?

pabender, 2024-09-13:
According to the documentation, while the z21 start ( the white version of the z21 ) doesn’t have a LocoNet connection, it does process LocoNet messages.

Back when this was written, there may not have been documentation on how to do the reads using the XpressNet reads.

I now have both a z21 Start and a Z21.  I’ll see if I can find time to look into what happens on both with the current firmware.

higaski, 2024-11-04:
Any news on this?

habazut, 2024-11-16:
If we can assume that the (white) z21 does not understand LocoNet over UDP (at least not completely) the JMRI should talk XPressNet over UDP instead, yes?

Can JRMI do that and just does not do it (for some unknown reason) or is that a capability that still needs to be made from scratch?

Regards....

bobjacobsen, 2024-11-17:
There some discussion that may be related in issue #13574

pabender, 2024-11-17:
> If we can assume that the (white) z21 does not understand LocoNet over UDP (at least not completely) the JMRI should talk XPressNet over UDP instead, yes?
> 
> Can JRMI do that and just does not do it (for some unknown reason) or is that a capability that still needs to be made from scratch?
> 

JMRI doesn't talk LocoNet over UDP or XPressNet over UDP to the Z21.

The Z21 has it's own protocol that encapsilates both the XPressNet and LocoNet messages.

I haven't had a chance yet to look at this with real hardware.  It's on my todo list.

pabender, 2024-11-18:
OK, so looking into this with an existing configuration and I couldn't replicate the problem, so I started looking at a few things (including issue #13574 which @bobjacobsen  mentioned might have some clues).

First, in the initial post on this issue, the programming track selection marked in the first image only impacts the service mode programmer.

To configure which connection provides the ops mode programmer, you have to go to the defaults in the preferences, which on creation of a new Z21 connection look like this:

[image]

With the default configuration, I do see behavior similar to what you see.  With the current firmware version for the (white) z21, I get the following sequence:

> LocoNet Tunnel Message: Byte Read on Main Track (Ops Mode): Decoder address 6195: CV1.
> 
>  LocoNet Tunnel Tx: LONG_ACK: Function not implemented, no reply will follow.

if the defaults are set to either the Z21XPressNet line or the Z21 line, then everything works as expected.

The reason for this behavior has to do with how the two subsystems (LocoNet and XPressNet) are initialized whichever one is initialized first (which is the LocoNet connection) is the one the defaults use as the initial default.  

If I recall correctly, the reason the connections are initialized in this order has to do with the dependency of the Ops Mode programmer on LocoNet for the readback in prior versions of the Z21 Firmware (because at one point there wasn't a way to get the results on the XPressNet subsystem).

pabender, 2024-11-18:
> If I recall correctly, the reason the connections are initialized in this order has to do with the dependency of the Ops Mode programmer on LocoNet for the readback in prior versions of the Z21 Firmware (because at one point there wasn't a way to get the results on the XPressNet subsystem).

Ok, so I remembered that partly correctly.  both the Black and Whtie Z21s respond to operation mode writes with a LocoNet response:

> XpressNet Tunnel Message: Operations Mode Programming Request: Byte Mode Write: 16 to CV 31 for Decoder Address 6195
>  LocoNet Tunnel Tx: Byte Write (No feedback) on Main Track: Decoder address 6195: CV31 value 16 (0x10, 00010000b).
> 
>  XpressNet Tunnel Message: Operations Mode Programming Request: Byte Mode Write: 5 to CV 32 for Decoder Address 6195
>  LocoNet Tunnel Tx: Byte Write (No feedback) on Main Track: Decoder address 6195: CV32 value 5 (0x05, 00000101b).
> 
>  XpressNet Tunnel Message: Operations Mode Programming Request: Byte Mode Verify: 5 to CV 490 for Decoder Address 6195
>  XpressNet Tunnel Reply: Programming Result CV: 490 Value: 0

(the above is  a read of CV 16.5.450 on an ESU Loksound V5 decoder, which requires writing two CVs before the read occurs ).

habazut, 2024-11-22:
To make this a **lot** easier for the unsuspecting **user** who probably knows much less about the capabilities of the differnet Z21/z21/z21start (_could one have chosen a more confusing way to name the products - probably not_) than we do,  JMRI could just **ask**  the Z21 of what type it is (via the XPressNet protocol) and then decide for the user what kind of _default_  is  the right one for the hardware in question. 

Regards,
Harald.

pabender, 2024-11-25:
> To make this a **lot** easier for the unsuspecting **user** who probably knows much less about the capabilities of the differnet Z21/z21/z21start (_could one have chosen a more confusing way to name [...]

Possibly, but that can cause issues when there are other systems connected to JMRI.

Except for the initial load, we make no assumptions about what defaults are correct after the user saves the preferences.