JMRI/JMRI #13386: Dispatcher issue auto allocate

fert75, 2024-08-27:
Dispatcher not work properly. I setting auto allocate option but only allocate one section "South main B25". If I setting allocate manually, the train start to run but "Auto restart" or "Reverse at end" not work. The Transit I'm using is " West to east main 1".
Please help.

[session.log](https://github.com/user-attachments/files/16764122/session.log)

[TEST7_FT-2.txt](https://github.com/user-attachments/files/16764121/TEST7_FT-2.txt)

dsand47, 2024-08-27:
You have a number of issues which are not caused by JMRI code.  These kinds of problems belong in the user group.

The session log file highlights some of the issues:

```
2024-08-27T11:50:58,669 startup.PerformFileModel              INFO  - Loading file C:\Users\FERTR\JMRI\My_JMRI_Railroad.jmri\TEST7_FT.xml [main]
2024-08-27T11:51:02,070 layoutEditor.BlockValueFile           INFO  - 1 of 1 block values restored. Rule=RESTOREONLYIFALLOCCUPIED, Power=OFF [AWT-EventQueue-0]
2024-08-27T11:51:06,999 dispatcher.DispatcherFrame            ERROR - No train in start block 'S24 (IB:AUTO:0009)', cannot create an Active Train [AWT-EventQueue-0]
2024-08-27T11:51:06,999 dispatcher.DispatcherFrame            WARN  - failed to create Active Train '' [AWT-EventQueue-0]
```

You are loading the xml during PanelPro startup.  You also have a transit configured to start a train as part of the startup process.  Since this is occurring before DCC-EX is fully functional, the sensor state has not been updated.

The references to section and transit names do not patch any names in the new xml file.

fert75, 2024-08-27:
Ok. Thanks. How can I solve this issues?

fert75, 2024-08-27:
[My_JMRI_Railroad.jmri.zip](https://github.com/user-attachments/files/16766487/My_JMRI_Railroad.jmri.zip)

dsand47, 2024-08-27:
You have two errors in the train info file.

You specified allocate "To Safe Secs" but the transit does not have any safe sections.  Change it to "Number of Sections:".

[image]

You are starting the train after loading.

[image]

You need to do things in discrete steps until you have gained experience.  Start PanelPro, manually load the layout config file, manually start a Dispatcher train.

dsand47, 2024-08-27:
After changing the allocation to 3 sections and removing the start on load, the simulated train goes around the loop repeatedly.

fert75, 2024-08-27:
Another question I would like to ask you is regarding occupation blocks. I have currently installed IF sensors. I would like to know if it is possible to emulate the occupation blocks with the IF sensors.
Later I will install Block occupancy. Could you recommend me a hardware that it works with DCC ex and JMRI?. Thank you so much!

dsand47, 2024-08-27:
> Another question I would like to ask you is regarding occupation blocks. I have currently installed IF sensors. I would like to know if it is possible to emulate the occupation blocks with the IF sensors. Later I will install Block occupancy.

Infra red sensors are "point" sensors.  For effective occupancy detection you should use "current detection" with resistive wheel sets on you your rolling stock.

> Could you recommend me a hardware that it works with DCC ex and JMRI?. Thank you so much!

That question needs to be taken to a DCC-EX support group.

fert75, 2024-08-29:
Finally I can solve the issues in transit "West to east main 1" but I have an issue in transit "Main 1 + crossover". This transit start at S24 block and go arround one lap to main track I, then go to the crossover (S23 to S29) to main track II, after the one lap there, exit in the crossover (S31 to 25) and start again (loop).
The issue is in the turnouts. Before starting Its turn "thrown" when should be "Close" to go straight ahead.

dsand47, 2024-08-29:
> I have an issue in transit "Main 1 + crossover". This transit start at S24 block and go arround one lap to main track I, then go to the crossover (S23 to S29) to main track II, after the one lap there, exit in the crossover (S31 to 25).

This does not describe the transit correctly.  It does one outer loop, takes the crossover to S30 and then back to the outer loop for a second time.  The transit ends in S24.

The JMRI simulation support for DCC-EX turnouts is minimal.  I have to use internal turnouts for testing.  You have several DCC-EX turnouts Inverted.  This can lead to unexpected results, especially if you are also using the "also throw" option.  I recommend that you change the switch machine connections so that JMRI Closed is the same as Layout Closed without using Inverted.

closed 2024-11-18 by github-actions[bot]