JMRI/JMRI #13390: 'Startup' options "hint" on GUI: add 'load "Panel" file _before_ starting any "Tables" tool'

devel-bobm, 2024-08-28:
Per jmriusers message 233855 ( [web link](https://groups.io/g/jmriusers/message/233855) ) and follow-on, a user had slow start-up times.  User was told that it could be the order of "startup" actions.  User got impressive speedup by opening the XML first, then opening various "Tables".

Of note, the user had this to say in jmriusers message 233879 ( [web link](https://groups.io/g/jmriusers/message/233879)) )

> I noticed that the preferences panel says "Scripts should be run last in most cases" - could we perhaps add "files should be loaded first" as well please?

For our purposes, I'd suggest something like "'Load table content and panels...', if used, should be loaded first.".

Or, "'Load table content and panels...', if used, should be loaded before any action to 'Open _any_ Table'."

devel-bobm, 2024-08-28:
A possible change to the bottom of he "Start Up" window:

[image]

danielb987, 2024-08-28:
Sounds very good. If you have already done the change in your branch, I suggest that you create a PR for it.

KenC57, 2024-08-28:
Just to be clear, is it having the tables open, or is this impacted by
just having the 'add button' for the tables in the startup?

devel-bobm, 2024-08-29:
@danielb987 , I will.  But I18N issues will exist; I only remember random bits of my high school French and Latin from way too many years ago!

danielb987, 2024-08-29:
> @danielb987 , I will. But I18N issues will exist; I only remember random bits of my high school French and Latin from way too many years ago!

Thank you!

@DanielBoudreau 
Can you assist @devel-bobm in French translations?

DanielBoudreau, 2024-08-29:
Yes.

Dan

bobjacobsen, 2024-09-13:
Closed by #13391

closed 2024-09-13 by bobjacobsen