JMRI/JMRI #13392: Communication problem between DecoderPro and Uhlenbrock USB to LocoNet Interface 63120

Railroad-Willi, 2024-08-29:
Previously I was using JMRI 5.6 which did work fine with the Uhlenbrock interface After updating to JMRI 5.8, DecoderPro did not communicate any more with the Uhlenbrock interface. Only way to get it back to working order was to downgrade JMRI to 5.6 again.
Hope you will be able to fix this issue in further releases.
Best regards
Harald

danielb987, 2024-08-29:
@Railroad-Willi 
Welcome to the JMRI project!

Start DecoderPro, go to menu item Help -> System Console. Click "Copy to clipboard" and paste it in a reply to this issue.

Railroad-Willi, 2024-08-29:
Hello Daniel,

this is the output of the JMRI 5.8 installation but from a secondary setup which does have no connection to the layout. As I wanted to continue working with DecoderPro I already did downgrade the JMRI on the PC connected to the layout.

12:24:52,190 apps.util.Log4JUtil                   INFO  - ****** JMRI log ******* [main]
12:24:52,221 apps.util.Log4JUtil                   INFO  - This log is stored in file: C:\\Users\\Harald\JMRI\log\session.log [main]
12:24:52,221 apps.util.Log4JUtil                   INFO  - This log is appended to file: C:\\Users\\Harald\JMRI\log\messages.log [main]
12:24:52,253 apps.AppsBase                         INFO  - DecoderPro version 5.8+Rbc21ce2ce7 starts under Java 11.0.20 on Windows 10 amd64 v10.0 at Thu Aug 29 12:24:52 CEST 2024 [main]
12:24:52,440 apps.gui3.Apps3                       INFO  - Starting with profile Meine_JMRI_Bahngesellschaft.3f5ec438 [main]
12:24:52,659 jmri.util.node.NodeIdentity           INFO  - Using 78cbb038-e205-443e-92d5-da2f0436e655 as the JMRI storage identity for profile id 3f5ec438 [AWT-EventQueue-0]
12:24:53,237 jmri.jmrit.roster.Roster              INFO  - Reading roster file with rootFromName(C:\Users\Harald\JMRI\Meine_JMRI_Bahngesellschaft.jmri\roster.xml) [main]
12:24:53,768 jmri.util.FileUtilSupport             INFO  - File path program: is C:\Program Files (x86)\JMRI\ [main]
12:24:53,768 jmri.util.FileUtilSupport             INFO  - File path preference: is C:\Users\Harald\JMRI\Meine_JMRI_Bahngesellschaft.jmri\ [main]
12:24:53,768 jmri.util.FileUtilSupport             INFO  - File path profile: is C:\Users\Harald\JMRI\Meine_JMRI_Bahngesellschaft.jmri\ [main]
12:24:53,768 jmri.util.FileUtilSupport             INFO  - File path settings: is C:\\Users\\Harald\JMRI\ [main]
12:24:53,768 jmri.util.FileUtilSupport             INFO  - File path home: is C:\\Users\\Harald\ [main]
12:24:53,768 jmri.util.FileUtilSupport             INFO  - File path scripts: is C:\Program Files (x86)\JMRI\jython\ [main]
12:24:54,252 mri.jmrit.throttle.PowerManagerButton INFO  - No power manager instance found, panel not active [main]

danielb987, 2024-08-29:
@Railroad-Willi 
Thanks. Can you do the same from the computer that is connected to the layout? This will give us information on how you have set up the connection.

danielb987, 2024-08-30:
@bobjacobsen 
Could you take a look at this issue? After upgrading to JMRI 5.8, DecoderPro failed to communicate with LocoNet.
Could it be a problem with the new serial library or some setting that's wrong?

bobjacobsen, 2024-08-30:
@Railroad-Willi
Sorry for the trouble!  

What do you have selected in your preferences for "System connection" and "Command Station type"?  What kind of hardware do you have for LocoNet Interface and command station?

Thank you!

Railroad-Willi, 2024-08-30:
Good afternoon Bob,

this is what I did select:

[cid:9bef93f4-2405-4b03-a085-3fce6f454dad]

I´m using an Uhlenbrock 63120 USB to LocoNet interface and an Intellibox I.
This does work with JMRI 5.6 but not with JMRI 5.8 or the latest test release.

Uhlenbrock told me that I might select Intellibox II as well as the interfaces are identical. I tried this on JMRI 5.8 and the latest test release but not with JMRI 5.6.

Best regards
Harald

bobjacobsen, 2024-08-31:
Unfortunately, the picture of your selections didn't come through.  Could you try that again please?  I don't think you can send pictures to GitHub via email, you probably need to add them via the web page: https://github.com/JMRI/JMRI/issues/13392

Railroad-Willi, 2024-08-31:
Settings for LocoNet are:

System manufacturer: Uhlenbrock
System connection: USB-LocoNet Interface 63120
Settings
Serial port: COM4
Command station type: Intellibox-I
Connection Prefix: L
Connection Name: LocoNet
Additional Connection Settings none

Best regards
Harald