JMRI/JMRI #13397: Implement circle colours for double slips in PanelPro

philguk, 2024-08-31:
It seems a historical decision was made NOT to support the circle colours option for closed/thrown switches when dealing with double slips

Could this decision be reviewed please as colouring the circles makes identification of the orientation of the switch positions much clearer 

And would make double switches consistent with other turnouts. At the moment, they behave inconsistently in this regard 

Thanks

danielb987, 2024-08-31:
@philguk 
Can you add a screenshot of what you mean? It makes it easier to understand your request.

dsand47, 2024-08-31:
> @philguk Can you add a screenshot of what you mean? It makes it easier to understand your request.

@danielb987 Look at the user thread at https://groups.io/g/jmriusers/message/233980

philguk, 2024-09-01:
> @philguk Can you add a screenshot of what you mean? It makes it easier to understand your request.

As well as Dave’s pointer (above), I’d like double slip turnout circles to behave the same as regular turnout circles if I set the option to display a different colo(u)r for Thrown vs Closed

closed 2024-09-03 by danielb987

philguk, 2024-09-03:
Awesome!

Closed #13397<https://github.com/JMRI/JMRI/issues/13397> as completed via #13414<https://github.com/JMRI/JMRI/pull/13414>.