JMRI/JMRI #13398: Confusing Route table behavior

dsand47, 2024-09-01:
A recent user group post, https://groups.io/g/jmriusers/message/233940, describes the unusual behavior of making Route table changes.

After editing a Route and clicking on **Update**, a blank **Add/Edit Route** detail window is presented instead of closing the **Edit Route** window.  The window has to be closed using the **Cancel Edit** button.

Prior to JMRI 4.19.7, the Add/Edit Route detail window included an Edit button.  By entering an existing system name and clicking on the Edit button, the Add/Edit Route detail window would load the route data and switch to edit mode.  By presenting an empty Add/Edit Route window after the Update, additional Routes can be edited or new ones created. 

PR #8554 refactored the Route table code.  The Edit button is no longer displayed and the supporting code appears to be gone.  However, the return to the blank Add/Edit Route window after an edit Update still occurs.

The proposed solution is to close the Edit detail window after processing the Update request.

philguk, 2024-09-01:
And for background, I think this refers to an issue I raised. I didn’t even realise it WAS a new route window. It just looks like the route I was editing but with the name gone

This is not how turnout (or sensor) edits work - it’s mightily confusing

bobjacobsen, 2024-09-02:
> The proposed solution is to close the Edit detail window after processing the Update request.

If that's all we need, I can work on that.

philguk, 2024-09-02:
iirc, that’s what  Edit Turnout and Edit Sensor do (I’m all for consistency)

Phil G

dsand47, 2024-09-02:
> > The proposed solution is to close the Edit detail window after processing the Update request.
> 
> If that's all we need, I can work on that.

I was planning on doing the change, but if you want to do the change, that is fine.  I have not started yet. The help pages also need to be updated.

closed 2024-09-03 by danielb987

philguk, 2024-09-03:
Thanks 👍

Closed #13398<https://github.com/JMRI/JMRI/issues/13398> as completed via #13411<https://github.com/JMRI/JMRI/pull/13411>.