JMRI/JMRI #13405: Question (or idea) for creating a new loco in DecoderPro

uli-2016, 2024-09-02:
Is it possible that the comment from the decoder file is also displayed when the decoder is selected, e.g. when the right mouse button is clicked or similar?
 
It would also be a good feature if this comment could also be included in the Decoder Comment field for the <new loco>.
 
[image]
[image]
[image]

Uli

bobjacobsen, 2024-09-02:
The JTree that displays the heirarchy doesn't have native support for tooltips, but there's some discussion about how to add them [here](https://stackoverflow.com/questions/272124/best-way-to-implement-tooltips-for-jtree).

bobjacobsen, 2024-09-02:
Looking more closely at it, tooltip support has already been added to `DecoderTreeNode` within `jmri.jmrit.symbolicprog.CombinedLocoSelTreePane`. It's already being used to put the CV8 value on manufacturer nodes. See line 248.  Lines 262, 276 and 290 are creating the model and family elements.  They're setting tooltips, but they don't seem to have the right content.

bobjacobsen, 2024-09-02:
PR #13413 has a solution for this.

PR #13413 (2024-09-03): Display decoder comments on the selection tree
Addresses and closes #13405
closed 2024-09-03 by danielb987