JMRI/JMRI #13408: Failing to install JMRI fresh on a particular computer previously running JMRI 4.24

jonasbjurel, 2024-09-02:
Hi, I have been fighting with my JMRI install since a few days now - without luck.
I have now come to a point where I have no good ideas to pursue and would need some help to find the trouble.
I have been running 4.24 for several years now without any problems, but 4 days ago it failed to start. The Java VM died almost immediately. My theory was that maybe a Windows update had messed around with my Java versions. I played back and forth with Java versions: JRE 8, 11, 21... without any luck. The next thing I did was to make a system restore to a point where I knew it was working, still no success. After that I uninstalled JMRI, renamed my configuration directory @ my home directory, made sure there was nothing JMRI related in my home\AppData directory. After that I made a fresh JMRI 5.8 install - still no luck, tried with JRE 11, 17 and 21 - no luck - the Java VM almost instantly crashes. Don't really know what to do next, installed 5.8 successfully on another machine - but that doesn't really help me I need it on this machine. And to reinstall this machine all together seems a bit drastic.
I have attached the java crashlog - it is a little verbose for my Java competence - but it is some access violation issue, and the garbage collector is mentioned quite alot?
Greatful for some help!
[hs_err_pid10816.log](https://github.com/user-attachments/files/16840478/hs_err_pid10816.log)

jonasbjurel, 2024-09-02:
It appears that later AMD Radeon graphic drivers don't play well with Java and the use-cases JMRI has for graphics.
As it appears, I have normally been using the NVIDIA graphical card, but as I'm now hooked up with HDMI the internal AMD graphics is used - causing the problem!
Downgraded to the generic Microsoft graphics driver Ver. 10.0.22621.1 and all is fine again.
I leave this one open - in case you want to issue a PR to the graphical libraries you are using. I leave it up to you how to handle it.

KenC57, 2024-09-02:
Jonas,

Here are a few ideas:

1. After installing Java, test it from a command window. Just do:

java -version

See if that works. If it doesn't, that usually points to poor
uninstalls of prior versions and things are getting mixed up. Special
tricks to get that cleaned up.

2. When installing JMRI make sure you pick the 'install for all users'
and let it use the defaults, that's the most likely to work.

3. If JMRI is sort of working, try making a whole new profile.
Sometimes data is saved in the profile that causes problems later.
This is the user preference issue. Specific steps can be taken once it
is suspected of being the issue.

-Ken Cameron, Member JMRI Dev Team
www.jmri.org
www.fingerlakeslivesteamers.org
www.cnymod.com

jonasbjurel, 2024-09-03:
Thank you, but it appears it was related to the AMD Radeon driver :-)