JMRI/JMRI #13422: Cannot instal latest JMRI on new laptop.

kelvinheads, 2024-09-05:


danielb987, 2024-09-05:
Which OS do you use? Windows, Linux or Mac?

Can you take a screen shot of the error message? Note that you cannot attach the screenshot to an email reply but instead you need to attach it to a reply on the GitHub web page.

kelvinheads, 2024-09-06:
Thank you. I have sorted this....kind of. The instructions say the 5.8 files are located in the 5.6 Github release but there is no sign of those files in the 5.6 release. And clicking on the "Download links" for 5.8 for Windows gets an error message. So I am running with 5.6 for the time being.
I am using Windows 11. I have been running DecoderPro for years on other laptops without issue, but cant seem to instal on the brand new laptop.
I cant seem to paste the message but it reads "5.8....normally isnt normally downloaded. Ensure you trust ....".
I hope this helps.

danielb987, 2024-09-06:
This message is shown by anti virus software, like the one built in into Windows. It's no error message, only a warning. It happens when a release is new and not many people have yet installed it

kelvinheads, 2024-09-06:
Thanks. And ow can we move on as it doesnt show the usual "Open File" in the Downloads page.

nigelcliffe, 2024-09-06:
The problem is that Windows security measures requires a large number of users to be using a program before they decide it is not suspect.   So, a file is marked as "suspect" until there is a large user base.  And getting that user base takes time. 

There is usually an option to override the Windows messages.   

In the browser downloads, it will say something like this (varies between browsers, this is Edge):

[image]

Clicking the " ... " at the right will bring up a dialog window, which after selecting the "show more" option will allow you to "keep" the file.  

That gets over the blocking of the download, and the the file into the "downloads".  

Then, on trying to run the file, Windows , it is possible that a further block with some text to click to reveal the option to say you really wish to run it.   (I've been unable to trigger that one on my machine just now).  

Third party antivirus software may add further blocks over those in Windows.   They are almost all operating in a similar way - a statistically low number of "its OK" reports from users meaning that the file is treated as "a risk".   And that will only change when the numbers of users gets larger.      For a piece of niche software in a relatively small hobby, the required number of users doesn't happen overnight.

crosland, 2024-09-06:
> Then, on trying to run the file, Windows , it is possible that a further block with some text to click to reveal the option to say you really wish to run it. (I've been unable to trigger that one on my machine just now).

Yes, this happens for me once when installing each new release. You can click "more info" to reveal a "run anyway" button.

[image]

It is possible to make this go away but it might involve paying Microsoft for some form of signing certificate. I did start a thread about this on one of the io groups some years ago but I can't find it at the moment.

closed 2024-12-02 by github-actions[bot]