JMRI/JMRI #13426: CDI Fetch Hangs if Configuration Tool is Opened.

crosland, 2024-09-06:
I first found this with my own node hardware but have found the same issue with a Tower-LCC, although the mechanism is slightly different. It could be a node issue, but two different code bases would suggest an issue in JMRI.

To re-create use the LCC Network Tree window to open the configuration dialog for the node. Whilst this is being fetched click on Configuration in the Supported Protocols branch.

The final messages with my hardware are:
JMRI receives end of multi-frame datagram:
[[1ce5812c] 27 33 32 27 3E 3C 6E 61]  R: (Middle of Datagram)
[[1de5812c] 6D 65 3E 45 76 65      ]  R: 02.01.2C.02.00.00 - 02.01.12.7F.02.D8 Datagram: (70) … <datagram contents snipped> 
JMRI start reading configuration 
[[1a12ce58] 20 80                  ]  S: 02.01.12.7F.02.D8 - 02.01.2C.02.00.00 Datagram: (2) 20.80
Node responds with OK
[[19a2812c] 0E 58 00               ]  R: 02.01.2C.02.00.00 - 02.01.12.7F.02.D8 DatagramReceivedOK with payload 00 flags 0x0
JMRI sends OK for datagram
[[19a28e58] 01 2C 00               ]  S: 02.01.12.7F.02.D8 - 02.01.2C.02.00.00 DatagramReceivedOK with payload 00 flags 0x0
JMRI receives a datagram for configuration tool
[[1ae5812c] 20 82 02 00 F0 FF FB   ]  R: 02.01.2C.02.00.00 - 02.01.12.7F.02.D8 Datagram: (7) 20.82.2.0.F0.FF.FB
JMRI sends OK
[[19a28e58] 01 2C 00               ]  S: 02.01.12.7F.02.D8 - 02.01.2C.02.00.00 DatagramReceivedOK with payload 00 flags 0x0

After that all traffic stops.

The Tower-LCC responds with a datagram rejected for the config tool datagram, but the traffic also stops after JMRI sends the datagram received OK:
[[1ca1b4eb] 75 6C 73 65 3C 2F 76 61]  R: (Middle of Datagram)
[[1da1b4eb] 6C 75 65 3E 3C 2F      ]  R: 02.01.57.00.05.9C - 02.01.12.7F.18.1C Datagram: (70) … <datagram contents snipped> 
[[1a4eba1b] 20 80                  ]  S: 02.01.12.7F.18.1C - 02.01.57.00.05.9C Datagram: (2) 20.80
[[19a484eb] 0A 1B 20 20            ]  R: 02.01.57.00.05.9C - 02.01.12.7F.18.1C DatagramRejected with payload 20 20 flags 0x2020
[[19a28a1b] 04 EB 00               ]  S: 02.01.12.7F.18.1C - 02.01.57.00.05.9C DatagramReceivedOK with payload 00 flags 0x0

The exact messages sequence in the console output can vary but, in both cases, the config tool display invalid memory spaces and the CDI display never populates.

Nodes appear to be responding correctly (i.e. they have not hung) if JMRI is quit and restarted.

System console output This is from a different run so JMRI node alias may differ):
16:34:31,668 enlcb.implementations.DatagramService ERROR - Overriding datagram transmit memo. old DatagramServiceTransmitMemo to 02.01.57.00.05.9C: 20.43.00.00.07.40.40 new DatagramServiceTransmitMemo to 02.01.57.00.05.9C: 20.80 [AWT-EventQueue-0]

bobjacobsen, 2024-09-06:
Thanks for the detail.  This does appear to be a bug in JMRI's queuing of memory requests.  I'll work on it, but it's likely to take a while as I'm away from hardware for the next 10 days.

crosland, 2024-09-06:
No problem, I'm just glad you can confirm it. It's not stopping me doing anything. I found it by accident, rather than any particular testing.

danielb987, 2024-10-21:
What's the status of this issue?

bobjacobsen, 2024-10-21:
On my list to fix, but somewhat down on the list.