JMRI/JMRI #13431: Can't open throttle windows

EdHowe, 2024-09-07:
I suddenly am unable to open throttle windows.  I had previously tried to create a consist but the consist screen locked up. After quitting and restarting the app, I've been unable to open throttle windows.  The below stack trace appears when I attempt to open a throttle.  I'm guessing there is a pref file with a bad value in it.  I'm guessing there is a settings file that has a bad value in it, so if someone could point me to it as a quick fix, I'd be grateful. 

13:58:32,146 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [AWT-EventQueue-0]
java.lang.IllegalArgumentException: Width (0) and height (19) cannot be <= 0
	at java.awt.image.DirectColorModel.createCompatibleWritableRaster(DirectColorModel.java:1016) ~[?:?]
	at java.awt.image.BufferedImage.<init>(BufferedImage.java:333) ~[?:?]
	at jmri.jmrit.consisttool.ConsistListCellRenderer.getListCellRendererComponent(ConsistListCellRenderer.java:58) ~[jmri.jar:?]
	at javax.swing.plaf.basic.BasicListUI.updateLayoutState(BasicListUI.java:1439) ~[?:?]
	at javax.swing.plaf.basic.BasicListUI.maybeUpdateLayoutState(BasicListUI.java:1389) ~[?:?]
	at javax.swing.plaf.basic.BasicListUI$Handler.valueChanged(BasicListUI.java:2785) ~[?:?]
	at javax.swing.DefaultListSelectionModel.fireValueChanged(DefaultListSelectionModel.java:219) ~[?:?]
	at javax.swing.DefaultListSelectionModel.fireValueChanged(DefaultListSelectionModel.java:199) ~[?:?]
	at javax.swing.DefaultListSelectionModel.fireValueChanged(DefaultListSelectionModel.java:246) ~[?:?]
	at javax.swing.DefaultListSelectionModel.insertIndexInterval(DefaultListSelectionModel.java:670) ~[?:?]
	at javax.swing.plaf.basic.BasicListUI$Handler.intervalAdded(BasicListUI.java:2743) ~[?:?]
	at javax.swing.AbstractListModel.fireIntervalAdded(AbstractListModel.java:153) ~[?:?]
	at javax.swing.DefaultComboBoxModel.addElement(DefaultComboBoxModel.java:130) ~[?:?]
	at javax.swing.JComboBox.addItem(JComboBox.java:716) ~[?:?]
	at jmri.jmrit.consisttool.ConsistComboBox.lambda$initConsistList$0(ConsistComboBox.java:65) ~[jmri.jar:?]
	at java.util.ArrayList.forEach(ArrayList.java:1540) ~[?:?]
	at jmri.jmrit.consisttool.ConsistComboBox.initConsistList(ConsistComboBox.java:65) ~[jmri.jar:?]
	at jmri.jmrit.consisttool.ConsistComboBox.notifyConsistListChanged(ConsistComboBox.java:91) ~[jmri.jar:?]
	at jmri.implementation.AbstractConsistManager.notifyConsistListChanged(AbstractConsistManager.java:150) ~[jmri.jar:?]
	at jmri.jmrit.consisttool.ConsistFile.consistFromXml(ConsistFile.java:92) ~[jmri.jar:?]
	at jmri.jmrit.consisttool.ConsistFile.readFile(ConsistFile.java:267) ~[jmri.jar:?]
	at jmri.jmrit.consisttool.ConsistFile.readFile(ConsistFile.java:239) ~[jmri.jar:?]
	at jmri.jmrit.consisttool.ConsistComboBox.init(ConsistComboBox.java:41) ~[jmri.jar:?]
	at jmri.jmrit.consisttool.ConsistComboBox.<init>(ConsistComboBox.java:30) ~[jmri.jar:?]
	at jmri.jmrit.throttle.AddressPanel.initGUI(AddressPanel.java:532) ~[jmri.jar:?]
	at jmri.jmrit.throttle.AddressPanel.<init>(AddressPanel.java:86) ~[jmri.jar:?]
	at jmri.jmrit.throttle.ThrottleFrame.initGUI(ThrottleFrame.java:358) ~[jmri.jar:?]
	at jmri.jmrit.throttle.ThrottleFrame.<init>(ThrottleFrame.java:107) ~[jmri.jar:?]
	at jmri.jmrit.throttle.ThrottleWindow.initGUI(ThrottleWindow.java:137) ~[jmri.jar:?]
	at jmri.jmrit.throttle.ThrottleWindow.<init>(ThrottleWindow.java:97) ~[jmri.jar:?]
	at jmri.jmrit.throttle.ThrottleFrameManager.createThrottleWindow(ThrottleFrameManager.java:55) ~[jmri.jar:?]
	at jmri.jmrit.throttle.ThrottleFrameManager.createThrottleFrame(ThrottleFrameManager.java:96) ~[jmri.jar:?]
	at jmri.jmrit.throttle.ThrottleFrameManager.createThrottleFrame(ThrottleFrameManager.java:86) ~[jmri.jar:?]
	at jmri.jmrit.roster.swing.RosterFrame.lambda$bottomRight$3(RosterFrame.java:280) ~[jmri.jar:?]
	at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:1967) ~[?:?]
	at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2308) ~[?:?]
	at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:405) ~[?:?]
	at javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:262) ~[?:?]
	at javax.swing.plaf.basic.BasicButtonListener.mouseReleased(BasicButtonListener.java:279) ~[?:?]
	at java.awt.Component.processMouseEvent(Component.java:6632) ~[?:?]
	at javax.swing.JComponent.processMouseEvent(JComponent.java:3342) ~[?:?]
	at java.awt.Component.processEvent(Component.java:6397) ~[?:?]
	at java.awt.Container.processEvent(Container.java:2263) ~[?:?]
	at java.awt.Component.dispatchEventImpl(Component.java:5008) ~[?:?]
	at java.awt.Container.dispatchEventImpl(Container.java:2321) ~[?:?]
	at java.awt.Component.dispatchEvent(Component.java:4840) ~[?:?]
	at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4918) ~[?:?]
	at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4547) ~[?:?]
	at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4488) ~[?:?]
	at java.awt.Container.dispatchEventImpl(Container.java:2307) ~[?:?]
	at java.awt.Window.dispatchEventImpl(Window.java:2772) ~[?:?]
	at java.awt.Component.dispatchEvent(Component.java:4840) ~[?:?]
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:772) ~[?:?]
	at java.awt.EventQueue$4.run(EventQueue.java:721) ~[?:?]
	at java.awt.EventQueue$4.run(EventQueue.java:715) ~[?:?]
	at java.security.AccessController.doPrivileged(Native Method) ~[?:?]
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85) ~[?:?]
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:95) ~[?:?]
	at java.awt.EventQueue$5.run(EventQueue.java:745) ~[?:?]
	at java.awt.EventQueue$5.run(EventQueue.java:743) ~[?:?]
	at java.security.AccessController.doPrivileged(Native Method) ~[?:?]
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85) ~[?:?]
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:742) ~[?:?]
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203) [?:?]
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124) [?:?]
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113) [?:?]
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109) [?:?]
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101) [?:?]
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:90) [?:?]

danielb987, 2024-09-07:
@EdHowe 
Welcome to the JMRI project!

@dsand47 
Which file should @EdHowe delete to reset the window positions? I have seen your answer before, but I don't remember it. Thank you.

dsand47, 2024-09-07:
Daniel,

The user-interface.xml file.  It is located at the "profile/<random directory name>" directory in the user files location.

Unfortunately the useful information at the beginning of the system console has been removed so the actual path is unknown.

It might also be a problem with the "consist.xml" file located at the "roster/consist" directory in the user files location.

Dave Sand

Lionel642, 2024-09-08:
I'm assigning this to myself to add a protector in   jmri.jmrit.consisttool.ConsistListCellRenderer.getListCellRendererComponent(ConsistListCellRenderer.java:58) ~[jmri.jar:?]

and avoid rescaling with a width of 0.

Meanwhile, a workaround might be to remove the icon associated with the roster entries in that consist, for some reasons, it tries to rescale that image icon to an image with a width of 0, which triggers an error. I guess the display size if very small, and hence creates that error.

closed 2024-09-09 by danielb987