JMRI/JMRI #13475: 5.9.4 download problem

Pensioner80, 2024-09-20:
Issue Type: __Bug__

every time I try to update to 5.8 or 5.9.34 I get report of it being unsafe. I have tried loading on several pcs with the same report

DecoderPro 5.6+R89a87446cb
Java HotSpot(TM) 64-Bit Server VM Oracle Corporation 11.0.20+9-LTS-256 mixed mode
Windows 11 10.0 amd64

dsand47, 2024-09-20:
This is not a JMRI bug.  It is a Windows process that checks for infrequently downloaded programs.

I will be creating an update to the Windows install help page to describe the process for Windows Defender.  Other anti-virus packages have similar processes.

The attached PDF is a preliminary mockup that should help you get an install to work.

[jmri install on windows.pdf](https://github.com/user-attachments/files/17076803/jmri.install.on.windows.pdf)

dsand47, 2024-09-24:
The Windows Install help page has been updated to include the steps to handle the Microsoft Defender untrusted download warning.

Pensioner80, 2024-09-24:
Thanks I'll let my club mates know.