JMRI/JMRI #13476: NCE Consist Tool Doesn't Work with USB Interface

toddthetrainnut, 2024-09-21:
As discussed in [this thread](https://groups.io/g/jmriusers/topic/nce_consisting_tool/27796458) over on the Users Groups.IO, the NCE Consist tool doesn't work when using the USB Interface. It looks like the tool uses commands that the USB interface doesn't support:

```
02:06:58,773 jmri.jmrix.nce.NceTrafficController   ERROR - attempt to send unsupported binary command to NCE USB: 8F [AWT-EventQueue-0]
jmri.JmriException: attempt to send unsupported binary command to NCE USB: 8F
	at jmri.jmrix.nce.NceMessageCheck.checkSerial_CMD(NceMessageCheck.java:97) ~[jmri.jar:?]
	at jmri.jmrix.nce.NceMessageCheck.checkMessage(NceMessageCheck.java:27) ~[jmri.jar:?]
	at jmri.jmrix.nce.NceTrafficController.sendNceMessage(NceTrafficController.java:555) ~[jmri.jar:?]
	at jmri.jmrix.nce.consist.NceConsistEditPanel.sendNceMessage(NceConsistEditPanel.java:1890) ~[jmri.jar:?]
	at jmri.jmrix.nce.consist.NceConsistEditPanel.readConsistMemory(NceConsistEditPanel.java:877) ~[jmri.jar:?]
	at jmri.jmrix.nce.consist.NceConsistEditPanel.getConsist(NceConsistEditPanel.java:786) ~[jmri.jar:?]
	at jmri.jmrix.nce.consist.NceConsistEditPanel.buttonActionPerformed(NceConsistEditPanel.java:516) ~[jmri.jar:?]
	at jmri.jmrix.nce.consist.NceConsistEditPanel$1.actionPerformed(NceConsistEditPanel.java:1995) ~[jmri.jar:?]
	at javax.swing.AbstractButton.fireActionPerformed(Unknown Source) ~[?:?]
	at javax.swing.AbstractButton$Handler.actionPerformed(Unknown Source) ~[?:?]
	at javax.swing.DefaultButtonModel.fireActionPerformed(Unknown Source) ~[?:?]
	at javax.swing.DefaultButtonModel.setPressed(Unknown Source) ~[?:?]
	at javax.swing.plaf.basic.BasicButtonListener.mouseReleased(Unknown Source) ~[?:?]
	at java.awt.AWTEventMulticaster.mouseReleased(Unknown Source) ~[?:?]
	at java.awt.Component.processMouseEvent(Unknown Source) ~[?:?]
	at javax.swing.JComponent.processMouseEvent(Unknown Source) ~[?:?]
	at java.awt.Component.processEvent(Unknown Source) ~[?:?]
	at java.awt.Container.processEvent(Unknown Source) ~[?:?]
	at java.awt.Component.dispatchEventImpl(Unknown Source) ~[?:?]
	at java.awt.Container.dispatchEventImpl(Unknown Source) ~[?:?]
	at java.awt.Component.dispatchEvent(Unknown Source) ~[?:?]
	at java.awt.LightweightDispatcher.retargetMouseEvent(Unknown Source) ~[?:?]
	at java.awt.LightweightDispatcher.processMouseEvent(Unknown Source) ~[?:?]
	at java.awt.LightweightDispatcher.dispatchEvent(Unknown Source) ~[?:?]
	at java.awt.Container.dispatchEventImpl(Unknown Source) ~[?:?]
	at java.awt.Window.dispatchEventImpl(Unknown Source) ~[?:?]
	at java.awt.Component.dispatchEvent(Unknown Source) ~[?:?]
	at java.awt.EventQueue.dispatchEventImpl(Unknown Source) ~[?:?]
	at java.awt.EventQueue$4.run(Unknown Source) ~[?:?]
	at java.awt.EventQueue$4.run(Unknown Source) ~[?:?]
	at java.security.AccessController.doPrivileged(Unknown Source) [?:?]
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source) ~[?:?]
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source) ~[?:?]
	at java.awt.EventQueue$5.run(Unknown Source) ~[?:?]
	at java.awt.EventQueue$5.run(Unknown Source) ~[?:?]
	at java.security.AccessController.doPrivileged(Unknown Source) [?:?]
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source) [?:?]
	at java.awt.EventQueue.dispatchEvent(Unknown Source) [?:?]
	at java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source) [?:?]
	at java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source) [?:?]
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source) [?:?]
	at java.awt.EventDispatchThread.pumpEvents(Unknown Source) [?:?]
	at java.awt.EventDispatchThread.pumpEvents(Unknown Source) [?:?]
	at java.awt.EventDispatchThread.run(Unknown Source) [?:?]
```

I'm able to reproduce this in the latest test release, 5.9.4. 

At the very least, I think we should disable the menu option for the Consist Tool if using the USB Interface since it isn't supported, but I'd really like this to work as it is provides an easy way to clear all consists from the system.

KenC57, 2024-09-21:
How is your connection configured? What jumper settings are on the NCE
USB? These things matter for the consist tool to work. Please post the
console log from your start of JMRI.

toddthetrainnut, 2024-09-21:
USB connection is configured as an SB5 @ 9600 baud. Despite NCE now recommending you always configure the board with no jumpers, I have jumper 3 in place to set the board to version 7.3.2 so that the programming track is disabled and so that I can change the cab # to be outside the 2-6 range reserved for actual throttles. I've set the USB Interface to cab 8.

[image]

This configuration appears to be working correctly - no errors in the PanelPro window, and Show Cabs works fine. 

[image]

[image]

I have also tried leaving all jumpers off and configuring the connection as a PowerCab, but this yields the same results; I can get the connection working and Show Cabs works fine, but the Consist Tool gives the same silent error as my first post.

Here's the console from startup with the connection configured as an SB5:

```
09:55:14,057 apps.util.Log4JUtil                   INFO  - ****** JMRI log ******* [main]
09:55:14,077 apps.util.Log4JUtil                   INFO  - This log is stored in file: C:\Users\toddt\JMRI\log\session.log [main]
09:55:14,078 apps.util.Log4JUtil                   INFO  - This log is appended to file: C:\Users\toddt\JMRI\log\messages.log [main]
09:55:14,103 apps.Apps                             INFO  - PanelPro version 5.9.4+R7726763089 starts under Java 17.0.12 on Windows 10 amd64 v10.0 at Sat Sep 21 09:55:14 CDT 2024 [main]
09:55:14,252 t.permission.DefaultPermissionManager INFO  - Permission file: C:\Users\toddt\JMRI\.permissions.xml [main]
09:55:14,264 t.permission.DefaultPermissionManager INFO  - Permission system is enabled: no [main]
09:55:14,280 apps.Apps                             INFO  - Starting with profile Central_Iowa_Spine_Line.3f47636c [main]
09:55:14,386 jmri.util.node.NodeIdentity           INFO  - Using 15c3490b-1e05-4a4e-b8c6-cc231af14d84 as the JMRI storage identity for profile id 3f47636c [AWT-EventQueue-0]
09:55:14,461 xml.AbstractSerialConnectionConfigXml INFO  - Starting to connect for "NCE" [main]
09:55:14,656 .jmrix.nce.usbdriver.UsbDriverAdapter INFO  - NCE: Connecting NCE USB to COM20 CP2102 USB to UART Bridge Controller [main]
09:55:14,660 .jmrix.nce.usbdriver.UsbDriverAdapter INFO  - NCE: Port NCE USB (COM20) opened at 9600 baud, sees DTR: true RTS: true DSR: false CTS: false DCD: false flow: NONE [main]
09:55:14,731 xml.AbstractSerialConnectionConfigXml INFO  - Starting to connect for "CAN" [main]
09:55:14,748 ers.gridconnect.GcSerialDriverAdapter INFO  - Connecting CAN to COM10 Simple CDC Device Demo [main]
09:55:14,752 ers.gridconnect.GcSerialDriverAdapter INFO  - LCC: Port USB Serial Port (COM10) opened at 57600 baud, sees DTR: true RTS: true DSR: false CTS: false DCD: false flow: NONE [main]
09:55:14,785 jmri.jmrix.nce.NceConnectionStatus    INFO  - NCE: NCE EPROM revision = 7.3.2 [AWT-EventQueue-0]
09:55:14,952 jmri.jmrix.nce.NceConnectionStatus    INFO  - NCE: Connected to NCE command station [AWT-EventQueue-0]
09:55:16,612 jmri.util.FileUtilSupport             INFO  - File path program: is C:\Program Files (x86)\JMRI\ [main]
09:55:16,612 jmri.util.FileUtilSupport             INFO  - File path preference: is C:\Users\toddt\JMRI\Central_Iowa_Spine_Line.jmri\ [main]
09:55:16,612 jmri.util.FileUtilSupport             INFO  - File path profile: is C:\Users\toddt\JMRI\Central_Iowa_Spine_Line.jmri\ [main]
09:55:16,613 jmri.util.FileUtilSupport             INFO  - File path settings: is C:\Users\toddt\JMRI\ [main]
09:55:16,613 jmri.util.FileUtilSupport             INFO  - File path home: is C:\Users\toddt\ [main]
09:55:16,613 jmri.util.FileUtilSupport             INFO  - File path scripts: is C:\Program Files (x86)\JMRI\jython\ [main]
09:55:17,670 apps.PanelPro.PanelPro                INFO  - Main initialization done [main]
```

KenC57, 2024-09-23:
I've confirmed that the PowerCab is using the memory access method that the normal command stations use and not the paged memory map method supported for the NCE-USB interface and PowerCab, SB5, and Twin. The log is showing it trying to use the 0x8F command when it should be using a series of 0xB3, 0xB4, and 0xB5 commands to perform the same operation.

A quick look at the code, I'm not seeing the consist interface code determining which memory access methods should be used for the operation. I thought it had been improved a few years ago so it could. I won't have much time to look into this during the week.

 Daniel Boudreau, do you recall what the status of this code should be doing?

KenC57, 2024-09-23:
I also checked and the macro code, which works in a similar way, works fine with the NCE-USB series systems. Either the consist code never got the right change or something broke and the consist tool isn't recognizing it is talking to a NCE-USB system and therefore uses the wrong commands to do the work.

DanielBoudreau, 2024-09-23:
Ken,

When I try to use the consist tool with USB version 7.3.0, I get this exception:

15:04:37,382 jmri.jmrix.nce.NceTrafficController   ERROR - attempt to send unsupported binary command to NCE USB: 8F [AWT-EventQueue-0]
jmri.JmriException: attempt to send unsupported binary command to NCE USB: 8F

Dan

KenC57, 2024-09-23:
The USB series don't support the 0x8F command. Recall that's a paged
memory map setup. So the 0xB4 and 0xB5 commands get us memory access.
And for reading, we can't read as many bytes at a time.

DanielBoudreau, 2024-09-24:
You never made the changes necessary to the NCE Consist tool.  I don't have the ProCab memory map, where can I get one?

Dan

KenC57, 2024-10-14:
I'm going to add the code, like used in the macro tool, so the consist tool will understand the PowerCab, SB5, Twin type systems. Since the space used by the consists in in the first 128 bytes of the memory page, it will work with the current firmware. A bug exists in the current firmware that prevents accessing the 2nd 128 bytes of the memory page.

DanielBoudreau, 2024-10-14:
Ken,

Have you figured out how to make the memory writes to cab memory permanent?

Dan

KenC57, 2024-10-14:
Dan,

One thing that Jim S confirmed is we should change the limit for the
macros as where it is writing for the upper half of the page is not a
good thing. I think that worked out to a limit or 7 or 8, I don't
recall if the limit works by number of macros or the macro number. I
recall they start at zero.

But right now I've started a branch to work on the consist editor and
plan to take a while with that. At least it is limited to the <128
offset, so won't have that issue. But before the next update, one of
us should change the limits in the macro to prevent going past the 127
offset. But since we just did a release, we've got a month before
needing to do that.

-ken c

KenC57, 2024-10-14:
I've not noticed a problem with the writes in the first half of the
pages. But tonight I'll do an explicit test of saving and power
cycling just to make sure. I suspect it might be an issue between 1.65
which you have and 1.65B which I have.