JMRI/JMRI #13480: Z21App crashes

aikopras, 2024-09-23:
At our model train club we are currently experimenting with the JMRI Z21-App. Unfortunately, the App crashes after a couple of hours, even with little or no use. This crash didn't happen just once, but basically every time the JMRI/Z21-App system has been running for a couple of hours. The exact period between system start and the Z21-App crash varies, however.

In the logfile I couldn't find any relation between the crash, and something that happened just before. In many cases there were several minutes between the crash, and the previous logfile item. 

The messages.log file includes the following entry:

System:
`2024-09-05T18:41:46,870 apps.Apps                             INFO  - PanelPro version 5.8+Rbc21ce2ce7 starts under Java 11.0.16.1 on Windows 10 amd64 v10.0 at Thu Sep 05 18:41:46 CEST 2024 [main]
`

Error message: 
```
2024-09-05T21:24:07,345 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [Z21 App Server]
java.util.ConcurrentModificationException: null
	at java.util.HashMap$HashIterator.nextNode(HashMap.java:1510) ~[?:?]
	at java.util.HashMap$ValueIterator.next(HashMap.java:1538) ~[?:?]
	at jmri.jmrit.z21server.ClientManager.handleExpiredClients(ClientManager.java:77) ~[jmri.jar:?]
	at jmri.jmrit.z21server.MainServer.run(MainServer.java:85) ~[jmri.jar:?]
	at java.lang.Thread.run(Thread.java:834) [?:?]
```

Aiko

danielb987, 2024-09-23:
@aikopras 
Welcome to the JMRI project!
Please upload the entire message.log file as an attachment. Note that you cannot do that by replying to an email. The only way to attach files is by the GitHub web page.

bobjacobsen, 2024-09-23:
Sorry for the trouble.  I'm working on this.

bobjacobsen, 2024-09-23:
Developer installers with (what I think is) a fix for this are here:

https://builds.jmri.org/jenkins/job/testreleases/job/bobjacobsen/

When you get the chance, could you try one of these to see if it fixes the problem?

It looks like the problem happens while handling a Z21 communications timeout.  Normally, there's a "heartbeat" message that will prevent this, but if the z21 server loses contact it will timeout and the bug will be encountered.

aikopras, 2024-09-23:
Hi
Thanks for the response.

> Please upload the entire message.log file as an attachment. 

I've attached it to my initial message

aikopras, 2024-09-23:
Dear Bob
> Developer installers with (what I think is) a fix for this are here:
> https://builds.jmri.org/jenkins/job/testreleases/job/bobjacobsen/

Wow. That's quick :-)
> When you get the chance, could you try one of these to see if it fixes the problem?

I hope we'll be able to test this this evening. However, since the error occurred only after some time, I assume we may need some longer period for testing. I'll report success (or failure), but since I will also be traveling this might take a week or so.

> It looks like the problem happens while handling a Z21 communications timeout. Normally, there's a "heartbeat" message that will prevent this, but if the z21 server loses contact it will timeout and the bug will be encountered.

This makes sense. At the end of the evening some people stop using the App, presumably in such a way that the LAN_LOGOFF message will not be send..

aikopras, 2024-09-24:
We've installed the new software yesterday evening without any problems. The first tests with Z21-Apps on phones worked well, but we need to run longer (several hours, several users) tests before we can conclude that there are no more crashes. I'll report after we've had these tests.
Thanks again for the quick patch! Highly appreciated!

danielb987, 2024-09-24:
@aikopras 
The fix by @bobjacobsen has been merged. I have reopened this issue so that you can report later when you have made more testing.

aikopras, 2024-10-11:
Hi
I promised to provide feedback regarding our further tests.
The good news is that the previous "crashes" of the Z21 software no longer occurred.
The bad news, however, is that we run into severe Loconet problems yesterday, during large scale testing. We were using a dozen of FREDs, an Intellibox and a PC running JMRI, connected via a [FremoLNShield](https://loconetovertcp.sourceforge.net/Interface/LocoLinx-UNO-FremoLNShield/#locolinxonarduinounowithfremolnshield) and USB. JMRI was used as Z21 server for 2 Iphones with the Roco Z21 App, but also as Loconet monitor, and was writing the Loconet messages to a logfile.

As said, we run into massive problems. Trains no longer listened to FREDs: FREDs had to be unplugged and replugged to get control again. There were differences between the iPhone Z21 Apps and the JMRI loconet monitor: the Z21 App on iPhones showed changes in train speed, although the user didn't touch the iPhone. The JMRI Loconet monitor and the FRED settings sometimes didn't match. 
We never had such problems in the past. Maybe JMRI was too busy with monitoring and couldn't handle the Z21 connections, maybe the LNShield got overloaded. We couldn't find any obvious issue in the logfiles, except for the fact that we had quite some Loconet traffic from the FREDs (many locs were configured for 126 speedsteps, which caused quite some traffic).

Only after disconnecting the JMRI PC and LNShield, we were able to drive trains with our FREDs again.

danielb987, 2024-10-11:
I don't know the problem, but LocoNet has an issue that it can be overloaded. If the traffic load is too high, LocoNet can break down. The issue is that if two LocoNet devices tries to send a message to the LocoNet at the same time, a collision occurs. Every collision takes some time to resolve. If there is a lot of traffic, the risk of collision goes much higher. And the more collisions you get, the slower the LocoNet bus becomes because the collisions slow the LocoNet down.

If the problem happens again, try to disconnect the Z21 apps from JMRI so that you only have JMRI and the FremoLNShield connected to the layout. No Z21 apps talking to JMRI. Then see if that solves the problem. If it doesn't, then disconnect the FremoLNShield on the LocoNet side. If it works now, try to reconnect the FremoLNShield to LocoNet. The purpose is to try to see what is causing the problem.

If the problem exists when JMRI and FremoLNShield is connected to LocoNet, but not when they are not, then try to disconnect the FREDs to see if that helps.

jy-roda, 2024-10-25:
Hi ! Thanks @danielb987 for fixing my mistakes.

I made the original contribution, and we use loconet (with digitrax) on our layout. 
Could you contact me so we can look into your issues ? We are fully jmri-based here (all throttles go trough JMRI before going to loconet)

closed 2025-08-04 by github-actions[bot]