JMRI/JMRI #13489: missing Zimo MX645

NikolausKlee, 2024-09-29:
Hi there,

i'm new to JMRI and until now I only use Decoder Pro. I really enjoy the easy options to Programm decoders, mostly from Zimo.

Her comes my question: I can't find the Zimo MX645 in the dropdown "installed decoder" and so can't do anything with that loco running with that decoder. I read the instructions for adding a decoder but they sound gibberish to me...
Is there any other help for my problem?

Best regards!
Nikolaus

danielb987, 2024-09-29:
Which version of JMRI do you use? The Zimo MX645 decoder have been included in JMRI the last 13 years. But maybe there is a new version of the decoder that JMRI doesn't know about?

@bobjacobsen 
Can you take a look at this? Thanks.

danielb987, 2024-09-29:
The instruction manual of Zimo MX645 is here:
https://www.zimo.at/web2010/documents/MX-small-decoders_EN.pdf

NikolausKlee, 2024-09-29:
Hi Daniel,
I use Version 5.8. 
I actually found what I did wrong though. I did not  understand the structure of the Zimo menu, but no I did and I'M feeling a bit silly now...

Thank you anyway for trying to help me.

danielb987, 2024-09-30:
Hi Nikolaus

No problem! You are welcome!

I'm closing this issue since the problem is now solved. If you have any more questions, feel free to open a new issue or ask on the [JMRI users list](https://groups.io/g/jmriusers/topics).

closed 2024-09-30 by danielb987