JMRI/JMRI #13498: LogixNg Local Variable informationis not saved by Store Table JMRI function

ErmannoV5, 2024-10-02:
Issue Type: __Bug__

The tables store function does not store/save the local variables defined in a LogixNG. All other information is stored however when the tables are accessed after restarting JMRI the local variables information in the LogixNG is lost.

The File>Store>Store All Tables Contents and Panels menu is used to store the LogixNG table.

The attached example shows the LogixNG when it is created and stored, and after JMRI is restarted.

DecoderPro 5.8+Rbc21ce2ce7
Java HotSpot(TM) 64-Bit Server VM Oracle Corporation 11.0.12+8-LTS-237 mixed mode
Windows 10 10.0 amd64

[IQ_AUTO_0024 LogixNg Example as created.txt](https://github.com/user-attachments/files/17218549/IQ_AUTO_0024.LogixNg.Example.as.created.txt)
[IQ_AUTO_0024 LogixNg Example after table Store and restarting JMRI.txt](https://github.com/user-attachments/files/17218553/IQ_AUTO_0024.LogixNg.Example.after.table.Store.and.restarting.JMRI.txt)
[systemInfo.txt](https://github.com/user-attachments/files/17218572/systemInfo.txt)

dsand47, 2024-10-02:
I have not looked at the code, but I suspect that local variables attached to the last line of a ConditionalNG are not retained since there is no reason to keep them since they will never be used.

Chapter 8 of the LogixNG reference talks about the scope of local variables.

I put my local variables on a **Many** statement so that they have a scope that includes the whole ConditionalNG.  For example:
```
       1:  LogixNG: Initializer
       2:      ConditionalNG: Startup Initialization ::: Startup
       3:          ! A
       4:              Many ::: IQDA:AUTO:0779 ::: Use default
       5:                 ::: Local variable "sensor", init to None ""
       6:                 ::: Local variable "turnout", init to None ""
       7:                  ! A1
```

I noticed that you are trying to do this using DecoderPro.  You should be using PanelPro for non-decoder activities.

ErmannoV5, 2024-10-02:
I have attached the full LogixNG for which I have the issue.  Please let me know if I have wrong. When I store it and restart JMRI the local variables disappear and the LogixNG aborts with errors.

I have also tried a small test using the PanelPro menu and I get similar results.

[Test-Set Ramp Speed 2.txt](https://github.com/user-attachments/files/17220210/Test-Set.Ramp.Speed.2.txt)

dsand47, 2024-10-02:
What does the **FRS Locomotive Ramp Down Table** csv file look like?

dsand47, 2024-10-02:
What are the errors when you restart PanelPro?  After starting PanelPro, go to **Help -> System Console** and click on the **Copy to clipboard** button.  Paste into a reply.

dsand47, 2024-10-02:
I suspect that one of the issues is the delayed execute actions.  Since you a using a set of delays, you will need to use individual timers.  The "Execute Delayed" queues each action for later execution.  The default is to use a single timer.  With "Reset on repeat", the last one wins. 

Here is a variation on your ConditionalNG.  I am using hard coded values instead of the table.  I also use the Execute Action.

```
LogixNG: Ramp
    ConditionalNG: Ramp ::: Startup
        ! A
            Many
               ::: Local variable "SelectedDCC", init to String ""
               ::: Local variable "SelectedSpeedStep", init to None ""
                ! A1
                    Set variable SelectedDCC to String "1234"
                ! A2
                    Execute A after 0 seconds. Ignore on repeat. Use individual timers
                        ! A
                            Many
                                ! A1
                                    Set variable SelectedSpeedStep to Floating number ".5"
                                ! A2
                                    Execute action SetThrottle
                ! A3
                    Execute A after 3 seconds. Ignore on repeat. Use individual timers
                        ! A
                            Many
                                ! A1
                                    Set variable SelectedSpeedStep to Floating number ".3"
                                ! A2
                                    Execute action SetThrottle
                ! A4
                    Execute A after 5 seconds. Ignore on repeat. Use individual timers
                        ! A
                            Many
                                ! A1
                                    Set variable SelectedSpeedStep to Floating number ".1"
                                ! A2
                                    Execute action SetThrottle
                ! A5
                    Return
                ! A6
                    // sub routine to set the throttle speed
                    Many ::: User Name: SetThrottle
                        ! A1
                            Throttle
                                ?~ Address
                                    Get local variable SelectedDCC as analog value
                                ?~ Speed
                                    Get local variable SelectedSpeedStep as analog value
                                ? Direction
                                    
                                ?~ Function
                                    
                                ? FunctionOnOff
                                    
                        ! A2
                            Log data: String.format: Throttle %s set to speed %s
```

The result:
```
11:01:50,173 jmri.jmrit.logixng.actions.LogData    WARN  - Throttle 1234 set to speed 0.5 [JMRI LogixNGThread]
11:01:53,179 jmri.jmrit.logixng.actions.LogData    WARN  - Throttle 1234 set to speed 0.3 [JMRI LogixNGThread]
11:01:55,177 jmri.jmrit.logixng.actions.LogData    WARN  - Throttle 1234 set to speed 0.1 [JMRI LogixNGThread]
```

ErmannoV5, 2024-10-03:
> What does the **FRS Locomotive Ramp Down Table** csv file look like?
Attached is the table CSV
[FRS_Locomotive_Speed_Ramp .txt](https://github.com/user-attachments/files/17237979/FRS_Locomotive_Speed_Ramp.txt)

ErmannoV5, 2024-10-03:
> What are the errors when you restart PanelPro? After starting PanelPro, go to **Help -> System Console** and click on the *

ErmannoV5, 2024-10-03:
> > What are the errors when you restart PanelPro? After starting PanelPro, go to **Help -> System Console** and click on the *

This is the System console log
[System Console 2024_10_03.txt](https://github.com/user-attachments/files/17238134/System.Console.2024_10_03.txt)

ErmannoV5, 2024-10-03:
> I suspect that one of the issues is the delayed execute actions. Since you a using a set of delays, you will need to use individual timers. The "Execute Delayed" queues each action for later [...]

Thank you for the suggestions  👍 I'm on a steep learning curve!

This part of a more complex routine.  Basically when a locomotive is  dected over a RFID sensors, it will rampdown the speed based on the table.  I'll rewrite the LogixNg based on your suggestions.  I'm perplexed by the Locol Variables going missing.  I'm going to use Global Variables for the moment but this will be an issues as I want the routine to execute across several sensors simultaneously.

Thanks, again.

dsand47, 2024-10-03:
There are over 200 errors in the system console listing.

To figure out what is actually happening, you need to upload the messages.log file located at **C:\Users\vitto\JMRI\log**.

You also need to upload the layout data xml file.  It will be located at **C:\Users\vitto\Documents\JMRI FRS Config\FRS Config Profile 1**.  Since that is an xml file, you need to change the suffix to .txt.

dsand47, 2024-10-03:
Note:  The missing local variables are merely a symptom of a much larger issue.

ErmannoV5, 2024-10-03:
> There are over 200 errors in the system console listing.
>
> To figure out what is actually happening, you need to upload the messages.log file located at **C:\Users\vitto\JMRI\log**.
>
> You also need to [...]

Attached as requested.  I could not identify the "layout data xml file", so I have attached all the xml files in the directory.

[messages.log](https://github.com/user-attachments/files/17238710/messages.log)
[session.log](https://github.com/user-attachments/files/17238712/session.log)

[dispatcheroptions.xml.txt](https://github.com/user-attachments/files/17238718/dispatcheroptions.xml.txt)
[FRS Config Turnouts.xml.txt](https://github.com/user-attachments/files/17238719/FRS.Config.Turnouts.xml.txt)
[monitorLog.txt](https://github.com/user-attachments/files/17238720/monitorLog.txt)
[ProfileConfig.xml.txt](https://github.com/user-attachments/files/17238721/ProfileConfig.xml.txt)
[roster.xml.txt](https://github.com/user-attachments/files/17238722/roster.xml.txt)
[UserPrefsProfileConfig.xml.txt](https://github.com/user-attachments/files/17238723/UserPrefsProfileConfig.xml.txt)
[blockvalues.xml.txt](https://github.com/user-attachments/files/17238724/blockvalues.xml.txt)
[catalogTrees.xml.txt](https://github.com/user-attachments/files/17238725/catalogTrees.xml.txt)
[decoderIndex.xml.txt](https://github.com/user-attachments/files/17238726/decoderIndex.xml.txt)
[DecoderProSpeedometer.xml.txt](https://github.com/user-attachments/files/17238727/DecoderProSpeedometer.xml.txt)

dsand47, 2024-10-03:
The layout data xml file is the file the you load after starting PanelPro.  It is the file selected using **File -> Load table content and panels**.

ErmannoV5, 2024-10-03:
> The layout data xml file is the file the you load after starting PanelPro. It is the file selected using **File -> Load table content and panels**.
This is the load table content and panels files which I load at startup in the preference settings.
[FRS_Panel_Operations V5.0.xml.txt](https://github.com/user-attachments/files/17239040/FRS_Panel_Operations.V5.0.xml.txt)

ErmannoV5, 2024-10-03:
I have made some further observations.  I have been accessing/editing the tables using "Listed Table Access" button on Decoderpro.  As a result of your advice I have now been using PanelPro.  Today I noticed the my ID Tags Table has less entries when I use PanelPro.   So my speculation is that layout data XML is corrupt.  Is there a way to "refresh" the XML Schema so that I do not lose the valid layout data?

dsand47, 2024-10-03:
I need the **session.log** file from a PanelPro session.

ErmannoV5, 2024-10-03:
Session log as requested. 

[session 2024_10_04.log](https://github.com/user-attachments/files/17250741/session.2024_10_04.log)

dsand47, 2024-10-03:
Try removing the file load from the startup actions and do a manual load after PanelPro is started.  

If you still have the "maleSocket is null" errors, do a store with a different file name.  Then stop, re-start and manually load the new file.

ErmannoV5, 2024-10-04:
Removed file load from startup and manulally loaded from Panelpro. Result: same messages ("Male Socket is Null").
The "Male Socket is Null" messgae only appears when the panel and table file is loaded.

Used PanelPro to Store file under a new file name, restarted and loaded manually the new file.  Result: same messages ("Male Socket is Null").   Session log attached.

[session 2024_10_04 1026.log](https://github.com/user-attachments/files/17251559/session.2024_10_04.1026.log)

dsand47, 2024-10-04:
It appears that you are going to need to find a backup version that is valid.  If you do the store with replace, the backup versions are in the backupPanels directory.

@danielb987 

The actual problem is that a store was done but actions and expressions do not have the MaleSocket element.  This creates hundreds of null MaleSocket errors.  And since local variables are attached to AbstractMaleSocket they also disappear after a restart and load.

danielb987, 2024-10-04:
@ErmannoV5 
Can you send me the files in the backupPanels directory?

My email address is:

db123 AT bergqvist.se

(replace AT with `@`)

dsand47, 2024-10-04:
> @ErmannoV5 Can you send me the files in the backupPanels directory?
> 
> My email address is:
> 
> db123 AT bergqvist.se
> 
> (replace AT with `@`)

Be careful what you ask for.  There can be hundreds of files.  I recommend getting a zip file of the backupPanels directory.

danielb987, 2024-10-04:
@dsand47 
Thanks. Yes, a zip filer would be good.

danielb987, 2024-10-04:
@ErmannoV5 
Can you reply here when you have sent me the backup files, preferably as a zip file? Or reply if you have a question.

ErmannoV5, 2024-10-04:
Attached is the panel backup directory (I have culled it down to 2024 to manage the  size of the file).

The "maleSocket is null" message first appears on the file **FRS_Panel_Operations V5.0_20240925170609.xml (modified ‎25-‎Sep-‎24 ‏‎3:54 PM)** .  Before that date no such message.

[FRS backupPanels-2024.zip](https://github.com/user-attachments/files/17264441/FRS.backupPanels-2024.zip)

Do you think that if I delete the LogixNGs that I modified/created on the 25-Sep-24 will make the problem go away?

danielb987, 2024-10-04:
What happens if you load the last file that doesn't have that message and then stores it? Will the stored file have that message if you open it?

ErmannoV5, 2024-10-05:
When I Store the file without the message and then restart and load the new stored file, the message **appears!**  I Also tried on a previous file without the message and have the same result - the message apperas.

Thus is apperas when I store a file without the message, when it is loaded the message appears.

danielb987, 2024-10-05:
> When I Store the file without the message and then restart and load the new stored file, the message **appears!**  

Thank you!

Can you upload both the file without the message and the file you get when you store the first file?
When you have stored the file, go to Help -> System console and click "Copy" and then paste it in a reply.

ErmannoV5, 2024-10-05:
File without Null message
[FRS_Panel_Operations V5.0_20240925155454.xml.txt](https://github.com/user-attachments/files/17264605/FRS_Panel_Operations.V5.0_20240925155454.xml.txt)
[Store of Before Message Log.txt](https://github.com/user-attachments/files/17264606/Store.of.Before.Message.Log.txt)

File with Null Message:

[FRS_Panel_Operations V5.0_20240925155454 Test Store.xml.txt](https://github.com/user-attachments/files/17264609/FRS_Panel_Operations.V5.0_20240925155454.Test.Store.xml.txt)
[Load of Stored Before and Null Message Log.txt](https://github.com/user-attachments/files/17264611/Load.of.Stored.Before.and.Null.Message.Log.txt)

For your information I upgraded to 5.8 on 6 Aug 2024.  I loaded the first backup after the upgrade.  I stored it and reloaded and the Null message appears.  This suggests the problem arose with 5.8.  However I dont understand that in between the upgrade and the Null message apperance I had several backups without the Null message.

danielb987, 2024-10-05:
@ErmannoV5 @dsand47 
I have found the bug but I need to thought thru on how to solve it. But there is a simple solution for @ErmannoV5 to get JMRI to work.

Start JMRI, go to `Edit -> Preferences`, select `LogixNG` and check `Install debugger`. Save settings and restart JMRI.

The bug is that the male sockets are not stored correctly when the LogixNG debugger isn't installed. I have to fix that, but it's probably not a problem to have it active.

dsand47, 2024-10-05:
> The bug is that the male sockets are not stored correctly when the LogixNG debugger isn't installed. I have to fix that, but it's probably not a problem to have it active.

That explains why I was able to load the bad file, store it and end up with a good file.  Of course the local variables were lost and need to be re-added.

Since the debugger is installed by default, no one else has reported the problem.

danielb987, 2024-10-05:
PR #13508 fixes this bug.

It also adds the `java/test/jmri/jmrit/logixng/configurexml2/` test suite that does LoadAndStore and StoreAndLoad with `Install debugger` unselected to test this.

ErmannoV5, 2024-10-06:
**Well done! Great effort!** 🥇 

I installed the debugger and stored the file - message disappeared.  It explains the behavior from my end.  I turned off the debugger as I did not use it.

I'm fine running with debugger.

Thank you so much for your effort, expertise and your promptness.

Ermanno.

danielb987, 2024-10-21:
Is this issue fixed? Or is it any more to do?

dsand47, 2024-10-21:
> Is this issue fixed? Or is it any more to do?

You fixed it with PR #13508.

danielb987, 2024-10-21:
> > Is this issue fixed? Or is it any more to do?
> 
> You fixed it with PR #13508.
> 

Thanks for the answer. The issue has a lot of comments so I asked to ensure that I didn't miss anything.

closed 2024-10-21 by danielb987