JMRI/JMRI #13502: OperationsPro HPT settings request

Breezlys, 2024-10-03:
I've been using OpsPro (and DecoderPro and PanelPro) for a long time and really love it, but I have an update request if it isn't too difficult to implement:

Would it be possible to add more granularity to the HPT setting, rather than accepting whole numbers only?

For example, 2.0, 2.1, 2.2, 2.3, ... 3.0, etc?
Or maybe 2.0, 2.25, 2.50. 2.75, 3.0, etc?

On my smallish layout, a whole number jump doesn't let me do much fine-tuning based on HPT. I suppose I could fib about my loco's HP ratings, but I feel that more granularity in the HPT settings would work better.

Thanks for your consideration!

danielb987, 2024-10-03:
@DanielBoudreau 
Is this something for you to take a look at?

DanielBoudreau, 2024-10-03:
The "Settings" HPT is used when the grade is 0.

The program when determining the required HPT for a grade uses the prototypical formula "HPT = Speed x % Grade / 12". The speed when determining the HPT is set to 36 MPH, which means that a 1% grade will require 3 HPT.

I've started the work needed to allow the settings HPT to have decimal places.  I've also made changes to allow the train speed to be modified for grades.

Breezlys, 2024-10-04:
Thanks, Dan!

I model a granger road in the midwest, so no real grades. This will be really helpful.

DanielBoudreau, 2024-10-04:
Changed in PR 13506

DanielBoudreau, 2024-10-05:
Development build is ready for you to test.  

https://builds.jmri.org/jenkins/job/Development/job/packages/

Breezlys, 2024-10-05:
Thanks again, Dan!

I have a busy schedule this weekend, so I probably won't be able to try it until Monday. I'll let you know how it goes.

Breezlys, 2024-10-10:
I was finally able to install and try JMRI.5.9.4plus+jenkins+20241009T2212Z+R12c1a037218. 

The decimal ability works great! I was able to build a train with several different values and saw the varying results in the build report. Now I just have to dial in the value I need to get the results I'm after. Thanks again, Dan.

I did, however, encounter a panel issue with the above-mentioned build. I'll open another issue for that.

DanielBoudreau, 2024-10-10:
Fixed with PR 13506.

closed 2024-10-10 by DanielBoudreau