JMRI/JMRI #13505: Cannot Connect MacBook running Decoder Pro to Digitrax PR4

JfinnessyTP, 2024-10-03:
am having a problem connecting a Digitrax PR4 Programmer to a Macbook.  I am wondering if this is a known problem, or if it has been solved.  The latest info I have found online is from 2021.
 
I am using a Macbook with a 1.3 Ghz i5 processor.  It has 8 GB memory and is running Ventura 13. 7. I am also trying to connect DecoderPro, JMRI Version 5.8+Rbc21ce2ce7 and Java 17.0.11.
 
I followed the instructions from the  Digitrax website  for connecting the PR3 to a Mac. (They indicated there was no difference between the PR3 and PR4 interface.)
 
When I try to connect, the lights on the PR4 are as follows:  Blue and Red are lit steadily, Green is mostly on but flashes about once per second.
 
Data from the System Report, Hardware, USB, USB2.1 Hub, VCOM PR4, shows the MAC has assigned the PR4 to Location ID 0x14131000/7, but this Location ID does not appear in the  Serial Port drop down menu in the DecoderPro  Connections =. Preferences Window.
 
(It’s also odd that when I actually go to the DecoderPro:  All Entries screen, in the lower left hand corner, it shows in green:  “Service Mode Programmer LocoNet is Online.” And Operations Mode Programmer Loconet is Online,” but it is not able to read any info from the Loco on the programming track.)
 
I am also not sure if this is related, but if I look at the IOUserSerial page from the System Report, it shows the following:
 
“Version 6.0.0
Last Modified:  9/5/24 3:11 AM
Bundle ID:  com.apple.driver.driverkit.serial
Notarized:  Unknown
Loaded:  No
Obtained from:  Not Signed
Location:  /System/Library/Extensions/IOUserSerial.kext
Validity Errors:
Validation Failures”. Kext has a CFBundleExecutable property but the executable can’t be found:  IOUserSerial
Signature Validation Errors:  Not Signed
Dependencies:    Incomplete
Signed by:  Not Signed”
 
Any help would be greatly appreciated!

dsand47, 2024-10-03:
Try following the steps at https://www.jmri.org/install/MacOSX.shtml#debug.

Dave Sand

JfinnessyTP, 2024-10-03:
Thank you!  That made the serial port show up in the terminal and then I was able to select it from the dropdown serial port in Decoder plus.But, I’m still having a problem.  The green light on the PR4 is now flashing on and off once per second.  When I try to read Loco info from the programming track, I get the following message:“programmer error:  programmer busy (302)”The light blinking as it is seems to indicate “USB Connected in Sound Programmer Mode” based on your list of Conditions for the LEDs.What is that and how do I get it to Program?Also, what do the black buttons do?  I can’t find documentation on them.Thanks.JudySent from my iPhone

closed 2024-12-23 by github-actions[bot]