JMRI/JMRI #13521: Panel display size differences in build JMRI.5.9.4plus+jenkins+20241009T2212Z+R12c1a037218 ?

Breezlys, 2024-10-10:
Just installed JMRI.5.9.4plus+jenkins+20241009T2212Z+R12c1a037218 to try out the HPT decimal value update that Dan did, and while that works fine, my panels don't display properly with that build. I reverted to 5.9.4 and the panels were once again correct, but when I reinstalled JMRI.5.9.4plus+jenkins+20241009T2212Z+R12c1a037218, my panels were wonky again.

I've hopefully attached .png screen captures of one of the panels. The wonky one is the one with the large white border.

[image]
[image]
[image]

danielb987, 2024-10-10:
@Breezlys 
Which panel type do you have? Layout editor, panel editor or control panel editor?

danielb987, 2024-10-10:
@Breezlys 
Do you have a tables and panels file that demonstrates the error? It's easier to track the bug down if we have a file showing the bug.

Breezlys, 2024-10-10:
There are actually four Panel Editor panels spread across two monitors. All of them exhibit the same symptoms, except the white border on the one full-screen panel is not as wide as on the smaller panels. Zip file of the panels and custom icons is attached.
[Panels.zip](https://github.com/user-attachments/files/17332893/Panels.zip)

danielb987, 2024-10-10:
Thank you. I have started a [thread](https://jmri-developers.groups.io/g/jmri/message/10458) on JMRI developers lis about this issue. Daniel Boudreau has [confirmed](https://jmri-developers.groups.io/g/jmri/message/10463) the problem.

danielb987, 2024-10-11:
Daniel Boudreau found the culprit of this problem and I think I have a solution.

If you have time and want to prove it, you find the installers here:
https://builds.jmri.org/jenkins/job/testreleases/job/danielb987/

danielb987, 2024-10-11:
The fix is included in the last test release JMRI 5.9.5.
Thanks for the bug report!

Breezlys, 2024-10-12:
Thanks for the quick resolution!

Unfortunately, I'm traveling and won't have access to the layout computer before this coming Wednesday. I will give 5.9.5 a try as soon as I get back home.

Breezlys, 2024-10-17:
Installed 5.9.5 this evening and all is well. Thanks again for the quick fix!

closed 2024-10-17 by Breezlys