JMRI/JMRI #13529: LE rendering circle

sidlo64, 2024-10-14:
If track segment changed from line to circle they are not rendered correctly. This happens after panel is loaded. After resizing panel the track segment is rendered correctly.
Tested on JMRI.5.9.5plus+jenkins+20241011T1613Z+R924ff3beeed.

[LEcircle.jmri.zip](https://github.com/user-attachments/files/17364978/LEcircle.jmri.zip)

[image]

dsand47, 2024-10-14:
The problem is that changing the track segment type or the angle of a circle does not recalculate the grid boundaries.

[image]

This probably also applies to the bezier and ellipse track segment types.

Moving an anchor point forces the grid update.

dsand47, 2024-10-17:
The panel size is set by `private Rectangle2D resizePanelBounds(boolean forceFlag)`.  The size is visible while in edit mode with the grid visible.  The size is recalculated whenever an object is added to the panel or an object has been moved.

Changes to object attributes, such as a track segment type changing to a circle, a track segment circle being flipped, the length of a turnout leg, etc., do not result in recalculating the panel size.  

Track segments don't have a position on a panel.  A track segment is the line drawn between two connections.  The connections are provided by turnouts, level crossings, anchor points, end bumpers and edge connectors.

It is possible to indirectly call `resizePanelBounds` by adding a public method to LayoutEditor.java.  However, the results are inconsistent due to how Layout Editor draws the panel.

danielb987, 2024-10-17:
> However, the results are inconsistent due to how Layout Editor draws the panel.

Can you give examples of that?

dsand47, 2024-10-18:
> > However, the results are inconsistent due to how Layout Editor draws the panel.
> 
> Can you give examples of that?

Here is an example movie.  It starts by moving an anchor point to get the grid correct.

https://www.dropbox.com/scl/fi/3tyqypo20glbgyl18siu4/flip-circle.mov?rlkey=h08g12hoz2hubzg28l794ee4x&dl=0

As you can see, the grid changes lag by one cycle of the change.  The other test case was using the change to the circle (arc) angle works, but generates a tremendous amount of overhead with a flickering screen when dragging the angle adjustment.

dsand47, 2024-10-18:
It now appears that we have been talking about the symptom of the actual problem.  When the panel is first displayed, the track segment data is incomplete which results in default locations of 0,0 and size of 50,50.

dsand47, 2024-10-20:
The circles (arcs) are truncated because the graphic "clip region" does not include the arc center points.

During the loading of a Layout Editor panel, the clip region is set by the location of panel items except for track segments.  Track segments don't have a location.  It is derived from the connection points such as turnouts and anchor points.

The center point of a track segment is half way between the end points.  For a track segment arc, the center point is not known until the arc is drawn.  This occurs after the clip region has been set. 

Unless someone wants to undertake a redesign of Layout Editor panel management, a simple workaround can be used.  Place a  "shape" point to the right and below the actual arcs that fall outside of the clip bounds.  This forces a larger clip region.

danielb987, 2024-10-21:
> Unless someone wants to undertake a redesign of Layout Editor panel management, a simple workaround can be used. Place a "shape" point to the right and below the actual arcs that fall outside of the clip bounds. This forces a larger clip region.

Sounds like a good idea.

Do you know where in the code that should be added? If so, do you have time to create a PR for it?

I don't have time for it at the moment.

dsand47, 2024-10-21:
> Sounds like a good idea.
>
> Do you know where in the code that should be added? If so, do you have time to create a PR for it?
>
> I don't have time for it at the moment.

Placing the shape point is a manual process when the user discovers that track segment arcs are being truncated.

dsand47, 2024-10-21:
There is another easy manual fix.  Split the 180º track segment into two 90º track segments.

closed 2024-10-24 by danielb987