JMRI/JMRI #13543: Simple programmer - options disappears

sidlo64, 2024-10-22:
Small bug with easy workaround.
Occurs in LocoNet and XpressNet connection. Maybe elsewhere.

Start Simple Programmer. Move to Ops programming.

In mode **Ops Byte** is **Short address** and **Long address** visible.

[image]

In mode **Ops Accessory Byte** is **Decoder address** and **Accessory address** visible.

[image]

In mode **Ops Accessory Extended Byte** is **Offset Address** visible.

[image]

Move back to mode **Ops Byte**.  Option **Long address** disappears.

[image]

Move back to mode **Ops Accessory Byte**. Option **Accessory address** disappears.

[image]

Workaround - close and open window Simple Programmer.

bobjacobsen, 2024-10-23:
What are you using as a system connection for this?  I get different options with my setup, and don't see the problem, so I'm probably using a different system connection.

sidlo64, 2024-10-24:
the above example is Linux + **XpressNet Simulator**
the example below is macOS + **LocoNet Simulator**
After click on **Ops Accessory Extended Byte** is an option Accessory address / Long address unavailable.

[image]