JMRI/JMRI #13548: XpressNet Monitor – bad address

sidlo64, 2024-10-23:
@pabender 
XpressNet Monitor icorrectly displays address for Ops mode programming for Accessory decoder and Extended Accessory decoder packet.

[image]

[image]

pabender, 2024-10-24:
Actually, XpressNet V3.5 to V3.8 only supports ops mode for mobile decoders.  Using ops mode for accessory decoders isn’t supported in these versions, so I can’t guarantee the simple programmer is sending the correct values to the decoders you are trying to program.

I haven’t checked to see if accessory decoder ops mode is supported in XpressNet V4.0, which JMRI doesn’t support yet.

sidlo64, 2024-10-24:
Regardless of the missing Lenz documentation, the XpressNet package is correct.
The Command station (e.g. NanoX) forwards wrapped packet to DCC as a correct DCC packet.
A decoder (e.g. NmraDcc arduino library) decodes it correctly.
Everything works as it should.

The problem is only in the XpressNet monitor message - text part.

Program XNetMessage.java call method LenzCommandStation.calcLocoAddress. It only correctly calculates the address for multifunction decoders.

It should be improved:
```
if getElement(2) == 0xxx xxxx || getElement(2) == 11xx xxxx then 
    multifunction decoder (loco)
else 
    if getElement(2) == 10xx xxxx && getElement(3) == 1xxx xxxx then 
        basic accessory decoder
    if getElement(2) == 10xx xxxx && getElement(3) == 0xxx xxxx then 
        extended accessory decoder
```
I'm not that good of a programmer so I don't want to write a fix.

pabender, 2024-10-24:
I’ll look into that, but my concern is more what does a Lenz command station ( or a Roco command station ) do with the instruction.  In the past, Lenz has forcibly removed what they consider edge case options from us.

The good news for you is I am working on some changes to the interpretation code that ends up putting the message into the monitor.  I was doing this as a part of some work that will add support for features of XPressNet 4.0.

sidlo64, 2024-11-20:
#13608

sidlo64, 2024-11-20:
#13608

closed 2024-11-20 by sidlo64