JMRI/JMRI #13574: Read type from decoder button is always disabled

chrisridd, 2024-11-04:
Issue Type: __Bug__

A few JMRI versions ago I was successfully able to use JMRI on my Mac with my white Roco z21 to automatically add locos to my roster using **New Loco** > **Read type from decoder**.

Now however, on a completely new installation (a new Mac), the **Read type from decoder** button is always disabled.

I can confirm Decoder Pro is able to communicate with the z21. The "Z21 Configuration Tool" window refreshes with plausible system information viz:
```
Hardware version: 0x204
Software version: 1.4300001
Serial number: 195444
```

I've tried JMRI 5.9.5+R924ff3beee as well as JMRI 5.8+Rbc21ce2ce7. I've tried openjdk 11 as well as openjdk 17.0.13 zulu.

I'm certain I'm doing something wrong, but it isn't clear what.

DecoderPro 5.9.5+R924ff3beee
OpenJDK 64-Bit Server VM Azul Systems, Inc. 17.0.13+11-LTS mixed mode, sharing
Mac OS X 15.1 aarch64

chrisridd, 2024-11-05:
Hi Bob,

I have a single connection defined. The white z21 doesn't have a separate programming track; everything is "Programming on Main" AIUI.

The "Read CV" button in the Single CV Programmer is disabled :-(

If I click Identify and watch Decoder Pro's Z21 Traffic, I do see it reading various CVs. The Identify process completes with an alert "2160 No roster entry found", and indeed the loco on the track has address 2160. When I "OK" that alert, all evidence of the found loco has gone. I was expecting it to be listed in the bottom of the main Decoder Pro window.

The Z21 traffic window log is here:
```
 LocoNet Tunnel Message: Byte Read in Direct Mode on Service Track: CV29.

 XpressNet Tunnel Reply: Broadcast: Service Mode Entry
 LocoNet Tunnel Tx: LONG_ACK: The Slot Write command was accepted.

 XpressNet Tunnel Reply: Broadcast: Normal Operations Resumed
 LocoNet Tunnel Tx: Set Global (Track) Power to 'ON'.

 LocoNet Tunnel Tx: Programming Response: Read Byte in Direct Mode on Service Track Was Successful: CV29 value 46 (0x2E, 00101110b).

 LocoNet Tunnel Message: Byte Read in Direct Mode on Service Track: CV17.

 XpressNet Tunnel Reply: Broadcast: Service Mode Entry
 LocoNet Tunnel Tx: LONG_ACK: The Slot Write command was accepted.

 XpressNet Tunnel Reply: Broadcast: Normal Operations Resumed
 LocoNet Tunnel Tx: Set Global (Track) Power to 'ON'.

 LocoNet Tunnel Tx: Programming Response: Read Byte in Direct Mode on Service Track Was Successful: CV17 value 200 (0xC8, 11001000b).

 LocoNet Tunnel Message: Byte Read in Direct Mode on Service Track: CV18.

 XpressNet Tunnel Reply: Broadcast: Service Mode Entry
 LocoNet Tunnel Tx: LONG_ACK: The Slot Write command was accepted.

 XpressNet Tunnel Reply: Broadcast: Normal Operations Resumed
 LocoNet Tunnel Tx: Set Global (Track) Power to 'ON'.

 LocoNet Tunnel Tx: Programming Response: Read Byte in Direct Mode on Service Track Was Successful: CV18 value 112 (0x70, 01110000b).

 LocoNet Tunnel Message: Byte Read in Direct Mode on Service Track: CV7.

 XpressNet Tunnel Reply: Broadcast: Service Mode Entry
 LocoNet Tunnel Tx: LONG_ACK: The Slot Write command was accepted.

 XpressNet Tunnel Reply: Broadcast: Normal Operations Resumed
 LocoNet Tunnel Tx: Set Global (Track) Power to 'ON'.

 LocoNet Tunnel Tx: Programming Response: Read Byte in Direct Mode on Service Track Was Successful: CV7 value 35 (0x23, 00100011b).

 LocoNet Tunnel Message: Byte Read in Direct Mode on Service Track: CV8.

 XpressNet Tunnel Reply: Broadcast: Service Mode Entry
 LocoNet Tunnel Tx: LONG_ACK: The Slot Write command was accepted.

 XpressNet Tunnel Reply: Broadcast: Normal Operations Resumed
 LocoNet Tunnel Tx: Set Global (Track) Power to 'ON'.

 LocoNet Tunnel Tx: Programming Response: Read Byte in Direct Mode on Service Track Was Successful: CV8 value 162 (0xA2, 10100010b).

 Z21 Serial Number Request
 Z21 Serial Number Reply.  Serial Number: 195,444
 Z21 Serial Number Request
 Z21 Serial Number Reply.  Serial Number: 195,444
 Z21 Serial Number Request
 Z21 Serial Number Reply.  Serial Number: 195,444
 Z21 Serial Number Request
 Z21 Serial Number Reply.  Serial Number: 195,444
 Z21 Serial Number Request
 Z21 Serial Number Reply.  Serial Number: 195,444
```

bobjacobsen, 2024-11-05:
I think you misunderstand what “Identify Loco” is supposed to do.  It reads the address from the decoder, and then selects an _existing_ roster entry that has the same addressing.  It’s not meant to create a new roster entry, and it doesn’t do that.

Bob

> On Nov 5, 2024, at 12:15 PM, Chris Ridd ***@***.***> wrote:
> 
> 
> Hi Bob,
> I have a single connection defined. The white z21 doesn't have a separate programming track; everything is "Programming on Main" AIUI.
> The "Read CV" button in the Single CV Programmer is disabled :-(
> If I click Identify and watch Decoder Pro's Z21 Traffic, I do see it reading various CVs. The Identify process completes with an alert "2160 No roster entry found", and indeed the loco on the track has address 2160. When I "OK" that alert, all evidence of the found loco has gone. I was expecting it to be listed in the bottom of the main Decoder Pro window.

—
Bob Jacobsen
***@***.***

bobjacobsen, 2024-11-05:
"New Loco"'s "Read type from decoder" can't work with programming on the main. Programming on the main operations require the address for the loco being read.  When doing "New Loco", that address isn't known yet.  I'd be very surprised if you were ever able to do that with programming on the main.  Is it possible that you used to use a programming track?

chrisridd, 2024-11-06:
So I went all the way back to the 4.5.16 release and Java 8, and the "Read type from decoder" button was enabled. Also the read button in "Single CV Programmer" was enabled and everything seemed to work as I remembered.

Going back through the official releases, the last one that worked as described above was JMRI.5.1.2+Ra3b171c8ba.

The next JMRI.5.1.3+R0f2da29e92 release has disabled buttons as I originally reported. There's nothing obviously related to the Z21 in the 5.1.3 release notes.

Hopefully that helps!

bobjacobsen, 2024-11-07:
Thank you for tracking that down!

I don't see the "Single CV Programmer" read button active in JMRI 5.1.2.  

I'm probably not configuring my test setup the same way as you are.  Could you provide the contents of your connection configuration for the Z21?  I've set "Roco", "Roco Z21" and a local server IP address.  

On the Single CV Programmer panel, it doesn't seem to matter whether I select "Z21Loconet" or "Z21XpressNet". Neither one enables the read button. Which do you use?

danielb987, 2024-11-08:
@bobjacobsen 
Which Z21 are you using?
- Z21 simulator 
- black Z22
- white Z21

The black Z21 has more features than the white one.

chrisridd, 2024-11-08:
@bobjacobsen Here are the details for the connection to my white z21:
```
System manufacturer: Roco
System connection: Roco Z21
IP address/hostname: 10.0.1.38
Connection prefix: Z
Connection name: Z21
```
Hostnames don't seem to work, maybe another bug.

My z21 is unlocked, hence (?) being able to give it a real IP address on my LAN.

For the "Simple CV Programmer" dialog, "Z21XpressNet" (the default) was selected in both "Programming Track" and "Program on Main" sections. I read CV 8 to test.

I could read CVs when Settings > Programming was set to either "Programming Track" OR "Programming on Main".

bobjacobsen, 2024-11-08:
I see the disabled buttons on both 5.1.2 and 5.1.3 (and the current code) when the Settings -> Defaults has Service Programmer set to Z21LocoNet.

I see the buttons enabled on both of those releases when Settings -> Defaults has Service Programmer set to Z21XpressNet.

I'm not recreating the difference you see between 5.1.2 and 5.1.3.  And with that default set, I'm seeing the buttons enabled even on the current code.  This is a surprise to me, because I think I understood you to say that your Z21 doesn't have a programming track.  (Note:  I don't have any Z21 hardware, of any model, to actually test with; I'm just running this to a blank port on a server)

So now I'm really confused.  

What do you have the Default for Service Mode Programmer set to?

chrisridd, 2024-11-08:
I'm using whatever the setup wizard created, ie Z21LocoNet is used for Throttles, Power Control, Command Station, Service Programmer, Ops Mode Programmer, and Consists.

I don't think I've ever changed them; I've no idea what they might be doing.

You're correct about the white z21 only having one set of track connections.

bobjacobsen, 2024-11-08:
Thanks.

Could you please try setting the Service Programmer to Z21XpressNet, save, restart and see if that makes a difference?

chrisridd, 2024-11-08:
Wow, that makes everything work!

pabender, 2024-11-18:
> So changing the setup as described is a good workaround for me.
> 
> But it sounds like something needs to change as a result of this issue; whether that is to update the setup wizard to create a different configuration for at least the white z21 models (not sure about the black Z21 models), or whether the change that stopped at least the white z21 from being able to read decoders was slightly wrong and needs fixing up.

I put some more details about why this is the way it is in response to issue #13381.  

This is a duplicate of that issue, so I'm going to close it as such.

closed 2024-11-18 by pabender