JMRI/JMRI #13592: Add "copy to clipboard" to XpressNet Monitor window

philguk, 2024-11-09:
I've been trying to assist in debugging some XpressNet oddities, and with fast changing messages it can prove tricky to actually select and copy 

The JMRI System Console window has a "copy to clipboard" button

It would be REALLY HELPFUL to have one on the XpressNet Monitor window too

Thanks

dsand47, 2024-11-09:
The monitor windows have an indirect copy to clipboard.  Click in the window, do a ctrl-A, ctrl-C.  (cmd for macOS).

You can also direct the output to a log file.

philguk, 2024-11-09:
> The monitor windows have an indirect copy to clipboard. Click in the window, do a ctrl-A, ctrl-C. (cmd for macOS). You can also direct the output to a log file.
> […](#)

iirc, when I did a Ctrl-A the next message to drop on the end of the log cleared the selection - so I needed to time my click+ctrl-A+ctrl-C VERY carefully

closed 2024-11-12 by icklesteve

philguk, 2024-11-12:
Cheers Steve

Phil G

Closed #13592<https://github.com/JMRI/JMRI/issues/13592> as completed via #13594<https://github.com/JMRI/JMRI/pull/13594>.