JMRI/JMRI #13606: Threading Model needs revisited

pabender, 2024-11-17:
I've got a problem that I've seen for a while and I'm just reporting.

I've got a system set up for speed matching that runs on a Raspberry pi.  It has an SPROG-PI and a Bachrus speedometer connected and we're running the SPROG-PI in command station mode so we can do POM programming.

If I do the following:
Start DecoderPro
Turn Track power ON
Open an ESU Loksound 5 decoder definition

Then I get the following popup window
[image]

and track power is turned off.

I can also make this happen by moving windows around on the screen.  The file loading was just the first (and simplest) case where I saw this occur.

In either case, I can dismiss the window and turn track power back on.

Also, I have seen similar issues (though not the popup ) on the Digitrax system at the club when loading a Loksound 5 decoder definition.  In that case we communication to the layout stops and are forced to restart JMRI to recover.

I know the problem here relates to efforts a few years ago that put most operations on a single thread.  That was done with 2 methods, one to put things on the GUI thread and one to put them on the Layout thread with the intention that we knew one day a single thread wasn't sufficient.  

I think it's time to split those threads so the layout operations can continue even when IO operations, such as loading a file or moving windows on the screen, are occurring.

danielb987, 2024-11-18:
> I think it's time to split those threads so the layout operations can continue even when IO operations, such as loading a file or moving windows on the screen, are occurring.

I think it's a HUGE step to make it work and I think that we are nowhere near to be able to make it happen soon.

@bobjacobsen 
- Is this even possible without breaking users Jython scripts? For example a script that calls `turnout.getState()` or `turnout.setState()` from the GUI thread?
- Is there a way to require annotation of classes or methods with [@InvokeOnGuiThread](https://jmri.org/JavaDoc/doc/jmri/InvokeOnGuiThread.html), [@InvokeOnLayoutThread](https://jmri.org/JavaDoc/doc/jmri/InvokeOnLayoutThread.html) or [@InvokeOnAnyThread](https://jmri.org/JavaDoc/doc/jmri/InvokeOnAnyThread.html)? And to check that a method that is expected to run on one thread doesn't directly call a method that's expected to run on another thread?

LogixNG is executed on it's own threads so it's designed to handle this already. But it's not annotated with a InvokeOnLogixNGThread.

bobjacobsen, 2024-11-19:
To answer your specific questions first:

> Is this even possible without breaking users Jython scripts? For example a script that calls turnout.getState() or turnout.setState() from the GUI thread?

I think so.  But solving that is part of the work that needs to be done.

Basically, we'd have to find the places where execution would transfer from one thread to another, and build that in automatically.  That means making things like `.setState()` do the thread transfer automatically.

I agree that'll be a lot of work.

> Is there a way to require annotation of classes or methods with @InvokeOnGuiThread, @InvokeOnLayoutThread or @InvokeOnAnyThread? 

There probably is.  Java has a lot of pre- and post-processor tools that could be adapted to that.  Maybe a custom SpotBugs plugin?  But JMRI currently has about 9851 classes.  That annotation is going to be a lot of work.

> And to check that a method that is expected to run on one thread doesn't directly call a method that's expected to run on another thread?

There is a `ThreadingUtil` method that can check for that at runtime, but it would have to be added in a bunch of places.

Now let me address the larger problem:  JMRI does too much atomic work on the AWT/Swing thread.  That needs to be broken up so that the AWT/Swing thread isn't blocked for so long.

A starting place would be the code that creates the DecoderPro panes: Construction of each pane could be enclosed in a `SwingUtilities.invokeAndWait` call (*).  Or, even better, perhaps the creation of each Variable could be wrapped like that, to give even more granularity.

(*) We should probably provide this via a method in the `ThreadingUtil` class.  The existing `ThreadingUtil#runOnGUI` method isn't it, because it runs immediately if already on the AWT/Swing GUI thread.

Pugwash1, 2024-11-19:
I was always taught that the writing of info to the screen, sending info to the task was a process that you communicated to via a pipe and any work was done on its  own thread. 
So anything in jmri that takes more than a milli second should spawn a thread.
I never really understood why property change listeners weren't spawning hundreds of threads.
Just my 1 cent.
Steve G.

danielb987, 2024-11-19:
I agree with @Pugwash1.

> A starting place would be the code that creates the DecoderPro panes: Construction of each pane could be enclosed in a `SwingUtilities.invokeAndWait` call (*). Or, even better, perhaps the creation of each Variable could be wrapped like that, to give even more granularity.

It seems to me that the solution is to create a new thread that constructs the panes. That thread will then call `SwingUtilities.invokeAndWait`for the creation of each Variable.

The window would need a message "Please wait while loading the CVs" or something like that.

crosland, 2024-11-19:
Paul, The popup is specific to the SPROG code, so you will not see it with any other system.

Which Pi-SPROG do you have? If it's a Pi-SPROG One with the original firmware then it would be beneficial to upgrade to Pi-SPROG 3 firmware, whic uses a much more robust serial protocol (actually gridconnect, the same as CBUS and LCC) and you should not see the error.

The loco stack is held in the Pi-SPROG 3 rather than in JMRI as is the case with original SPROGs and Pi-SPROG One. This means that JMRI is not required to create and send all of the DCC packets in a timely fashion.

I think the speed of the SD card can also have an impact as the Pi has a flash filing system.

Reducing logging as much as possible also helps in that respect.

You could also try adjusting SprogConstants.CS_REPLY_TIMEOUT but it's already set to 2.5 s. See also SprogConstants.PACKET_DELAY_WARN_THRESHOLD which control logging of delays. I think you can override these in startup script.

pabender, 2024-11-21:
> Paul, The popup is specific to the SPROG code, so you will not see it with any other system.
> 

yes, I knew that.  If I hadn’t also seen similar issues on a LocoNet based system, then a localized solution might be appropriate.

> Which Pi-SPROG do you have? If it's a Pi-SPROG One with the original firmware then it would be beneficial to upgrade to Pi-SPROG 3 firmware, whic uses a much more robust serial protocol (actually gridconnect, the same as CBUS and LCC) and you should not see the error.

If is a PI-SPROG 1, i think.

> I think the speed of the SD card can also have an impact as the Pi has a flash filing system.
> 

The other machine i saw issues on ( with LocoNet ) is a Windows laptop.  This isn’t an SPROG or PI specific issue.

pabender, 2024-11-21:
> I was always taught that the writing of info to the screen, sending info to the task was a process that you communicated to via a pipe and any work was done on its own thread. So anything in jmri [...]

100% agree, and that is more like things were done before we added the methods to put things on the layout thread vs the GUI thread.

danielb987, 2025-01-13:
@bobjacobsen 
> A starting place would be the code that creates the DecoderPro panes: Construction of each pane could be enclosed in a SwingUtilities.invokeAndWait call (*). Or, even better, perhaps the creation of each Variable could be wrapped like that, to give even more granularity.

Can you explain how you mean with this? I don't understand how this could solve the problem.
If the swing thread calls an "invokeAndWait()" method, it would still be blocked.

bobjacobsen, 2025-01-13:
> Can you explain how you mean with this? I don't understand how this could solve the problem.
> If the swing thread calls an "invokeAndWait()" method, it would still be blocked.

Wa a bad idea, didn't work.

bobjacobsen, 2025-11-12:
> The timeout talking... issue can occur during operations if you, e.g., drag a throttle window around the screen. The serial comms thread is starved in some way whilst the GUI activity takes place.

Maybe this is system-specific. AFAIK e.g. the C/MRI polling continues even when the Swing/AWT thread is blocked and e.g. LocoNet continues to receive and queue (though not to fully process) while blocked.  

I'll see if I can recreate this with the SPROG connection.

closed 2026-02-09 by github-actions[bot]

pabender, 2026-04-18:
@danielb987 

This is the last bit of the SPROG log before it fails.  

```
19:33:17.311: cmd: "O FF 00 FF"
19:33:17.411: rep: "P> "
19:33:17.411: cmd: "O FF 00 FF"
19:33:19.912: cmd: "-"
19:33:20.014: cmd: "-"
19:33:20.084: rep: "P> "
19:33:20.114: cmd: "-"
19:33:20.114: rep: "P> "
19:33:20.314: rep: "P> " 
```
And this is the RFID log 

```
19:33:19.318: msg:  Reply from CORE-ID reader. Tag read A2009100D9 checksum EA valid? yes
19:33:21.079: msg:  Reply from CORE-ID reader. Tag read A2009100D9 checksum EA valid? yes
```

Remember that the SPROG code is causing the SPROG to repeat values on the track, so that 1.5 seconds without a response after the command at 19:33:17.411  is no good at all.

I was able to get it to do the same thing by a series of window open and closes, but not something as reproduce able as either the file load or the RFID read.

I'll also note that this is on my laptop, not on a Raspberry Pi as the original instance was.

bobjacobsen, 2026-04-19:
Looking at the overall pattern in the SPROG data, I would have expected a reply to the 2nd `O FF 00 FF` around 19:33:17.511 or so.  That seems to have gotten lost. That's well before the time that the RFID messages show for their reception.

After the missed reply to the 2nd `O FF 00 FF` there are three `-` commands and three `P>` replies, but they're not properly interleaved. 

It seems unlikely that the AWT/Swing thread was busy for 2.5+ seconds due to an RFID reception or moving a window. Rather, it looks to me like something went wrong within the SPROG command station code during a shorter busy time and a message was missed, resulting in the command-reply matching getting out of synch, followed by an unsuccessful attempt at a recovery.

But that doesn't explain how the message could have been missed so long before the RFID's indicated time.  Unless that somehow is blocking threads for a long time _before- the message is sent to its monitor. That doesn't seem likely.

crosland, 2026-04-19:
> From "Paul Bender" ***@***.***>
> To "JMRI/JMRI" ***@***.***>
> Cc "Andrew Crosland" ***@***.***>; "Mention"
> Date 19/04/2026 16:05:37
> Subject Re: [JMRI/JMRI] Threading Model needs revisited (Issue #13606)

I have now reviewed it and it looks OK.

Should I have had a separate notification from github that my review was 
required? Just wondering if there's a setting I need to tweak.

Andrew