JMRI/JMRI #13634: Stephen Blackstone’s website url is not active

beizenga, 2024-11-28:
Stephen has either moved or deleted his website. The hyperlink and related text requires adjustment.

Thanks

icklesteve, 2024-11-28:
Searching the project for the text "Blackstone" produced just 1 unrelated result,
https://github.com/JMRI/JMRI/blob/master/xml/decoders/SoundTraxx_Lighting_Athearn_Genesis_Caboose.xml#L27

Please provide an example URL.

beizenga, 2024-11-28:
Sorry Steve,

 

I thought the page was captured in context. Hopefully, the following is more helpful.

 

See this page https://www.jmri.org/help/en/html/hardware/arduino/index.shtmlthere . Under “Using multiple Arduino via RS-485” there is a reference to  https://www.motorhomesites.org.uk/railway/JMRI_Two_Arduino_control_with_CMRI.php

 

Be aware that on https://www.jmri.org/help/en/html/hardware/arduino/index.shtml#intro the same error occurs under “Emulating a CMRI node” when accessing https://www.motorhomesites.org.uk/railway/JMRI_Arduino_Setup.php

 

I expect these two errors are related. I am not certain that this I the same Blackstone as Blackstone Models. An online wander suggested that this might have been a business website that changed hands during COVID.

 

On a separate note, you need to know how much I appreciate the work that you and the crew are doing to keep the site and the IO Groups forum working. Please pass on my appreciation.

 

Happy Thanksgiving!

 

Brad

 

https://github.com/JMRI/JMRI/blob/master/xml/decoders/SoundTraxx_Lighting_Athearn_Genesis_Caboose.xml#L27

bobjacobsen, 2024-11-28:
You can retrieve archived versions of those URLs from 

https://web.archive.org/web/20220811002302/https://www.motorhomesites.org.uk/railway/JMRI_Two_Arduino_control_with_CMRI.php

https://web.archive.org/web/20220118050523/https://www.motorhomesites.org.uk/railway/JMRI_Arduino_Setup.php

The 2nd one has a lot of excess stuff at the top, but the real content looks OK.

Bob

> On Nov 28, 2024, at 12:32 PM, beizenga ***@***.***> wrote:
> 
> 
> Sorry Steve, 
> 
> 
> 
> I thought the page was captured in context. Hopefully, the following is more helpful. 
> 
> 
> 
> See this page https://www.jmri.org/help/en/html/hardware/arduino/index.shtmlthere . Under “Using multiple Arduino via RS-485” there is a reference to https://www.motorhomesites.org.uk/railway/JMRI_Two_Arduino_control_with_CMRI.php 
> 
> 
> 
> Be aware that on https://www.jmri.org/help/en/html/hardware/arduino/index.shtml#intro the same error occurs under “Emulating a CMRI node” when accessing https://www.motorhomesites.org.uk/railway/JMRI_Arduino_Setup.php 
> 
> 
> 
> I expect these two errors are related. I am not certain that this I the same Blackstone as Blackstone Models. An online wander suggested that this might have been a business website that changed hands during COVID. 
> 
> 
> 
> On a separate note, you need to know how much I appreciate the work that you and the crew are doing to keep the site and the IO Groups forum working. Please pass on my appreciation. 
> 
> 
> 
> Happy Thanksgiving! 
> 
> 
> 
> Brad 
> 
—
Bob Jacobsen
***@***.***

beizenga, 2024-11-28:
Thanks BobBrad

https://web.archive.org/web/20220811002302/https://www.motorhomesites.org.uk/railway/JMRI_Two_Arduino_control_with_CMRI.php

https://web.archive.org/web/20220118050523/https://www.motorhomesites.org.uk/railway/JMRI_Arduino_Setup.php

—

Bob Jacobsen

***@***.***

danielb987, 2024-11-29:
@bobjacobsen @icklesteve 

> You can retrieve archived versions of those URLs from
>
> https://web.archive.org/web/20220811002302/https://www.motorhomesites.org.uk/railway/JMRI_Two_Arduino_control_with_CMRI.php [...]

A suggestion is to change the links to the archive links and add a note that the links go to the archive and not the original page.

beizenga, 2024-11-30:
Bob,I found the archived files useful but I noticed that the image files containing the example screen shots are disconnected from the archive version. Thought you might want to be aware.Best regards,Brad

https://web.archive.org/web/20220811002302/https://www.motorhomesites.org.uk/railway/JMRI_Two_Arduino_control_with_CMRI.php

https://web.archive.org/web/20220118050523/https://www.motorhomesites.org.uk/railway/JMRI_Arduino_Setup.php

—

Bob Jacobsen

***@***.***

PR #13637 (2024-12-01): Update Arduino Brackstone support weblinks
Fixes #13634

Initially unable to locate as Stephen Brackstone is named in the support pages, not BlackStone.
closed 2024-12-01 by danielb987