JMRI/JMRI #13641: Dispatcher bypassed connection specified turnout delay interval

KenC57, 2024-12-01:
It seems that somehow Dispatcher isn't using a normal call for turnouts. In this case LocoNet, which has a turnout delay specified in the connection, that delay is not used when Dispatcher is issuing turnout commands. Here is a snap from the monitor from a user:

Take a look, with 600ms delay set in the connection settings, here is the sequence of turnout changes generated by a Route:
 
[image]

As you can see 600ms delay is inserted between every two commands.
 
And now Dispatcher with the same settings:
 
[image]

600ms delay setting was completely ignored... Just 31ms between the first two commands, and no more than 256ms between any other...

More background can be found in this thread:
https://groups.io/g/jmriusers/topic/109849243#msg236912

Pugwash1, 2024-12-01:
@KenC57 
Lets be very clear Dispatcher has never had since its inception the idea of delaying all turnout commands. 5 years ago someone added the functionality to the connection to delay all turnout commands on a connection and implemented for Routes and SignalMastMatrix, no where else. No one has asked for that functionality in dispatcher since then and noone seems to have had a problem in the last 15years.
This is not a bug.
That having been said I am working up a PR to add the ability  to Optional use the the delay in Dispatcher. 
This will be the _third_ place it is being used, I am not changing the 150+ other places it isnt, but could be used.

KenC57, 2024-12-01:
My view is that a connection level delay should delay all turnout
commands to that connection. While I'd agree that the usefulness of a
connection level delay has good and bad sides. But I see that the
Output Interval for LocoNet is indeed defined only for use by systems
like Routes and Output Matrix Signal Masts. But it would seem from
that clue that any subsystem that is able to generate large bursts of
commands should consider some sort of throttling method.

 

Our problem is that few users read the details that deeply to find out
the connection level Output Interval only applies to some subsystems
and not the connection in general. It had me fooled. Maybe the name of
the issue needs to change to reflect that Dispatcher needs to have a
way to throttle its system impact on volume of messages?

 

I'm seeing the value in some other connections but not seeing the
meaning defined in other connections. Is it general or just specific
to certain tools. Is it better to have that concept of delays at the
connection level or just in subsystems that can generate a volume of
messages?

 

Now would there be a better design? Likely the underlying need is that
some turnouts on a given connection need a delay between specific
turnouts. I'm thinking the case of the capacitive discharge driven
turnouts that need a recharge time as one case. Another would be any
case where a turnout driver can't take commands one right behind
another. Yes, using software to make up for what is more likely a
hardware limitation, but that's what we have done many times.

 

The problem with a turnout level delay is how to know when it would
apply or not. Is it that a given turnout needs X time of no turnout
messages before it is given or is it that another turnout message
shouldn't happen until X time after this one? Or would it have to
think about when you have a mix of no time delay and time delay
turnouts and when to include or exclude the time delay? This seems
pretty sticky concept to come up with a viable solution.

closed 2025-02-24 by github-actions[bot]