JMRI/JMRI #13668: JMRI, wants to program, not on mainline

djcalcutt, 2024-12-11:
When trying to program on the mainline, JMRI seem to want to use the programming track some times.

JMRI 5.9.6 using NCE (single com port 4) com for programming track is no being used at the moment. Second NCE is in the repair shop.

So while trying to program on the mainline, an error shows. JMRI has been set to using the one and only command station.

See video https://nas.cloudbusinessportal.ca/myMedia/mo/sharing/Vyj46U2e0

you will see what I'm talking about.

djcalcutt, 2024-12-15:
still having an issue of trying to program on the main line (JMRI wants to use the programming track, error 301)

I now have received my original commands stations back fro NCE. 
one dedicated for programming track, the other for the main line.

My JMRI is set up to use the main line for everything except for programming...

so, still when trying to program on the mainline, error 301 pops up.

KenC57, 2024-12-15:
DJ,

 

I take it you have two NCE connections in the profile. Then in the
Defaults in Preferences, you have one selected for everything except
Service Programmer, and that is the only selected one on the other
connection?

 

-Ken Cameron, Member JMRI Dev Team

www.jmri.org <http://www.jmri.org/> 

www.fingerlakeslivesteamers.org
<http://www.fingerlakeslivesteamers.org/> 

www.cnymod.com <http://www.cnymod.com/>

djcalcutt, 2024-12-15:
Yes. Ken… i’m the guy building the GUI with locomotive control. Ive talk to you on the phone. http://canadianlocomotivelogistics.ca:22080/web/webThrottle.html?loconame=1Yes frustrating. Dec 11, 2024 NCE | Synology Photosnas.cloudbusinessportal.ca Darrin J. CalcuttSent from my iPhone

 

 

<http://www.fingerlakeslivesteamers.org/>

dsand47, 2024-12-15:
Make sure that the "everything else" NCE command station is the first one in the connection list.  The "programming track" command station is second.

Dave Sand

djcalcutt, 2024-12-15:
Yes Darrin J. CalcuttSent from my iPhone

 

 

<http://www.fingerlakeslivesteamers.org/>

djcalcutt, 2024-12-16:
so today I’m trying to program a Bachman decoder that has an tsunami two Eco. You can see when hmri  is reading all the configuration variables. It sometimes has errors 301Darrin J. CalcuttSent from my iPhone

 

 

<http://www.fingerlakeslivesteamers.org/>

KenC57, 2024-12-16:
DJ,

 

I've seen occasional errors while doing long reads. Generally those
are reading the CV tab but then when done I go and scroll the CV list
and manually retry the few that didn't work.

 

I presume you noticed that regardless of the defaults, you can pick
which connection for the programming tools to use?

 

-Ken Cameron, Member JMRI Dev Team

www.jmri.org <http://www.jmri.org/> 

www.fingerlakeslivesteamers.org
<http://www.fingerlakeslivesteamers.org/> 

www.cnymod.com <http://www.cnymod.com/>

djcalcutt, 2024-12-16:
Yes exactly. Darrin J. CalcuttSent from my iPhone

 

 

 

<http://www.fingerlakeslivesteamers.org/>

djcalcutt, 2025-01-03:
I'm going to swap out a booster, command station and see if that changes things. 

[https://nas.cloudbusinessportal.ca/myMedia/mo/sharing/sGxkXoPoO](https://nas.cloudbusinessportal.ca/myMedia/mo/sharing/sGxkXoPoO)

djcalcutt, 2025-01-03:
By the way for testing purposes today, I am using usbport.com, three directly to the command station serial port nine pin

djcalcutt, 2025-01-04:
I'm also using Soundtraxx part 829002, which is a PTB – 100 programming track booster. 

I have an extra one so I might install it or remove the one I have and see if the problem still exist

djcalcutt, 2025-02-24:
Hmmm any developments on this?

djcalcutt, 2025-07-05:
Still waiting for a solution to the nagging problem.

closed 2025-11-03 by github-actions[bot]