JMRI/JMRI #13689: Audio Table Error

klk32003, 2024-12-17:
See [jmriusers](https://groups.io/g/jmriusers/topic/110139641) for details.

The error

```jmri.jmrit.beantable.ListedTableFrame ERROR - Error when adding jmri.jmrit.beantable.AudioTableAction to display [Listed Table Generation]```

occurs, when a column was hidden. The Audio Table is no longer be displayed. To solve the problem, the file ```user-interface.xml``` must be deleted or edited (just set the column to false again in tag ```<table xmlns="" name="jmri.jmrit.beantable.AudioTableAction">```).

JMRI 5.0 shows the same behavior.

I have looked at this, but to no avail. Advice is always welcome.

```
09:36:51,798 jmri.jmrit.beantable.ListedTableFrame ERROR - Error when adding jmri.jmrit.beantable.AudioTableAction to display [Listed Table Generation]
java.lang.ArrayIndexOutOfBoundsException: 5 >= 5
	at java.util.Vector.elementAt(Vector.java:466) ~[?:?]
	at javax.swing.table.DefaultTableColumnModel.getColumn(DefaultTableColumnModel.java:295) ~[?:?]
	at jmri.jmrit.beantable.BeanTableDataModel.setColumnToHoldButton(BeanTableDataModel.java:551) ~[classes/:?]
	at jmri.jmrit.beantable.AudioTableAction$AudioTableDataModel.configEditColumn(AudioTableAction.java:452) ~[classes/:?]
	at jmri.jmrit.beantable.AudioTablePanel.<init>(AudioTablePanel.java:83) ~[classes/:?]
	at jmri.jmrit.beantable.AudioTableAction.createModel(AudioTableAction.java:134) ~[classes/:?]
	at jmri.jmrit.beantable.AudioTableAction.getPanel(AudioTableAction.java:139) ~[classes/:?]
	at jmri.jmrit.beantable.AudioTableAction.getPanel(AudioTableAction.java:1) ~[classes/:?]
	at jmri.jmrit.beantable.ListedTableFrame$TabbedTableItem.addPanelModel(ListedTableFrame.java:449) ~[classes/:?]
	at jmri.jmrit.beantable.ListedTableFrame$TabbedTableItem.<init>(ListedTableFrame.java:397) ~[classes/:?]
	at jmri.jmrit.beantable.ListedTableFrame.initComponents(ListedTableFrame.java:128) [classes/:?]
	at jmri.jmrit.beantable.ListedTableAction$1.run(ListedTableAction.java:69) [classes/:?]
	at java.lang.Thread.run(Thread.java:840) [?:?]
```

bobjacobsen, 2024-12-17:
Don't have time to work on this right now, but I think this is one of those cases where you have to convert between the column numbers in the view and those in the model.  See

https://docs.oracle.com/javase/8/docs/api/javax/swing/JTable.html#convertColumnIndexToView-int-

https://docs.oracle.com/javase/8/docs/api/javax/swing/JTable.html#convertColumnIndexToModel-int-

I'm not sure whether a fix should be applied in `jmri.jmrit.beantable.BeanTableDataModel` or `jmri.jmrit.beantable.AudioTableAction`.  If this happens with lots of table types, then it probably should be in `BeanTableDataModel`.  If it's specific to the Audio table, then it should go in `AudioTableAction`.

bobjacobsen, 2024-12-17:
Thinking about this a little more, the real problem might be that the `setColumnToHoldButton` call is coming too late, after the preferences have been loaded and the columns are hidden.  A cleaner solution might be to move that call earlier in the creation of the Audio Table pane.

klk32003, 2024-12-17:
Thank you! This is a good starting point.

PR #13694 (2024-12-20): fix AudioTable Panel hidden columns
Sets Edit button columns before initialising tables.
Fixes #13689