JMRI/JMRI #13697: Sections - Memory Leak when Editing?

ssellmaker, 2024-12-20:
Issue Type: __Bug__

JMRI lags and becomes unresponsive when attempting to edit Sections made in PanelPro. JMRI must be forced quit with the Windows Task Manager. The only Section I can edit is System Name IY:0001, all others cause JMRI to become unresponsive. Windows Task Manager reported about 400 MB of memory usage for JMRI when operating normally, after trying to edit one of the Sections, this will increase to about 2000 MB, this is why I believe it is a memory leak. Updating JMRI and Java from 11 to 17 did not resolve the issue. If it is an issue with my file, please let me know what I did wrong as I did not encounter any errors in JMRI. I am happy to provide any additional files or information if needed. 

DecoderPro 5.10+Rca461bd266
OpenJDK 64-Bit Server VM Azul Systems, Inc. 17.0.13+11-LTS mixed mode, sharing
Windows 10 10.0 amd64
[profile.zip](https://github.com/user-attachments/files/18214968/profile.zip)
[session.log](https://github.com/user-attachments/files/18214969/session.log)
[systemInfo.txt](https://github.com/user-attachments/files/18214970/systemInfo.txt)

ssellmaker, 2024-12-20:
[Layout with Block detection and 2 IP cameras timer and automation V5.zip](https://github.com/user-attachments/files/18215024/Layout.with.Block.detection.and.2.IP.cameras.timer.and.automation.V5.zip)

The profile folder did not seem to upload correctly so I re uploaded it, I also included my layout file.

dsand47, 2024-12-20:
I recommend that you run PanelPro directly instead of starting DecoderPro and then switching to PanelPro.  See if that makes a difference.

ssellmaker, 2024-12-20:
No difference when running PanelPro directly unfortunately.

dsand47, 2024-12-20:
I see the same issue.  However, if I change Preferences -> Display from Nimbus to Metal, the problem does not occur.

ssellmaker, 2024-12-20:
Brilliant!!! This solved the issue, thank you so much for the quick assistance!! It looks like I will be getting used to the Metal theme now :)

KenC57, 2024-12-20:
I have seen many things within JMRI that work better, or only work,
using Metal vs any of the other L&F choices. On Windows systems, I
don't recall what I use on Linux.  It's on my list of things I always
setup when I do a new installation.

Pugwash1, 2025-02-10:
This will be fixed with #13858 . Its the same issue as GTK+ has. You must never update an exclude to exclude the currently selected item in a JComboBox.