JMRI/JMRI #13699: Programmer ignores modeType OPSACCEXTBYTEMODE

sidlo64, 2024-12-21:
Programmer ignores modeType `OPSACCEXTBYTEMODE` in decoder definition file. It will use `OPSBYTEMODE` instead.

How to reproduce:
With roster entry OPSACCEXTBYTEMODE 
address 501
CV 700
Value 100
Write changes

With LocoNet connection 
is sent
`[EF 0E 7C 67 00 03 75 00 21 3B 64 7F 7F 0D]  Byte Write (No feedback) on Main Track: Decoder address 501: CV700 value 100 (0x64, 01100100b).`
it should be
`[ED 0B 7F 51 0D 3D 61 6E 3B 64 57]  Extended Accessory Decoder CV Write: Address 501 CV 700 with 100.`

With XpressNet connection
is sent
`[packet:E6 30 C1 F5 EE BB 64 D3]   Operations Mode Programming Request: Byte Mode Write: 100 to CV 700 for Decoder Address 501`
it should be
`[packet:E6 30 BD 61 EE BB 64 3B]   Operations Mode Programming Request: Byte Mode Write: 100 to CV 700 for Decoder Address 501`

[image]

[OpsAccExtByteMode.jmri.zip](https://github.com/user-attachments/files/18218758/OpsAccExtByteMode.jmri.zip)

[decoders.xml.zip](https://github.com/user-attachments/files/18218761/decoders.xml.zip)

bobjacobsen, 2024-12-21:
Can't dig into this right now, but if somebody wants to look at it I suggest starting with `java/src/jmri/jmrit/progsupport/ProgOpsModePane.java` line 260:

```
        // add OPSACCBYTEMODE & OPSACCEXTBYTEMODE if possible
        if (modes.contains(ProgrammingMode.OPSBYTEMODE)) {
            if (!modes.contains(ProgrammingMode.OPSACCBYTEMODE)) {
                log.debug("   adding button for {} via AccessoryOpsModeProgrammerFacade", ProgrammingMode.OPSACCBYTEMODE);
                modes.add(ProgrammingMode.OPSACCBYTEMODE);
            }
            if (!modes.contains(ProgrammingMode.OPSACCEXTBYTEMODE)) {
                log.debug("   adding button for {} via AccessoryOpsModeProgrammerFacade", ProgrammingMode.OPSACCEXTBYTEMODE);
                modes.add(ProgrammingMode.OPSACCEXTBYTEMODE);
            }
        }
```

It looks like it'll only add `OPSACCEXTBYTEMODE` if `OPSBYTEMODE` is already present.  Is that the case here?

sidlo64, 2024-12-21:
No. Decoder definition Public_Domain_sidlo_OPSACCEXTBYTEMODE.xml has written:
```
<programming>
    <mode>DIRECTBYTEMODE</mode>
    <mode>OPSACCEXTBYTEMODE</mode>
</programming>
```
One type for service mode and one type for operation mode.

devel-bobm, 2024-12-21:
Have you defined 'ops="yes"' in the '<programming' statement?  I _assume_ that the `<mode>` modifies how the `ops="yes"` feature operates...

There is one decoder which implements `OPSACCEXTBYTEMODE`": the `NCE_Light-It_signal.xml`.  It defines itself as:
```
        <programming direct="no" paged="no" register="no" ops="yes">
            <capability>
                <name>Ops Mode Accessory Programming</name>
                <parameter name="Address Type">signal</parameter>
                <parameter name="Delay">500</parameter>
            </capability>
            <mode>OPSACCEXTBYTEMODE</mode>
        </programming>
```

My Digitrax configuration seems to generate the extended accessory CV Write accesses.  (But I do not have hardware to test...)

sidlo64, 2024-12-21:
Thank you. I used:
```
<programming direct="yes" paged="no" register="no" ops="yes">
    <capability>
        <name>Ops Mode Accessory Programming</name>
        <parameter name="Address Type">signal</parameter>
        <parameter name="Delay">500</parameter>
        </capability>
    <mode>OPSACCEXTBYTEMODE</mode>
</programming>
```    
Now it works in service mode and operation mode too.
Help page is here [https://www.jmri.org/help/en/html/apps/DecoderPro/CreateDecoderAdvanced.shtml#accessoryops](url)
I will close this issue.

closed 2024-12-22 by sidlo64