JMRI/JMRI #13700: Specify ordering of functions/function labels

philguk, 2024-12-21:
The WiThrottle protocol (and/or the devices that use it) always display decoder functions in FUNCTION NUMBER order (F0, F1, F2 ..... etc)

This means that to change the order of function labels in a throttle to support operational or user considerations the decoder must be reprogrammed to "move the functions around". Often this can take considerable effort and may even not be possible

It would alleviate this contraint if the SEQUENCE of the functions could be separated from the actual function button numbers. Obviously the ordering could still default to the way it is now but specifying a function SEQUENCE would be much simpler than reprograming decoders

I imagine that this would require changes to both JMRI and to any clients using the WiThrottle protocol - unless JMRI could somehow "hide" the reordering of labels from the clients

Lionel642, 2025-03-19:
I'll put this in my backlog

closed 2026-10-20 by Lionel642

philguk, 2026-10-20:
> requests on the go with GitHub Mobile for iOS
> and Android
> Download it today!

Interesting (and thanks)

Where can I see details of what's been changed/added??

Phil G

Lionel642, 2026-10-20:
Heelo, 
see PR #15514 , online documentation is already updated, see bottom of [this page](https://www.jmri.org/help/en/html/tools/throttle/RostersMediaPane.shtml). A test release should come soon with this feature. Please note this is only implemented in JMRI desktop throttles for now, not in Wiithrottle or Enginedriver.

philguk, 2026-10-20:
AMAZING work

Especially like

“ A roster entry can also be dropped here, all function entries will be copied from that roster entry.

A row from the function labels editor of another roster entry can be dropped as well, all properties will be copied to the function where it is dropped, except for the function number that won't be updated”