JMRI/JMRI #13706: Ghost/Adhoc Help Changes

dsand47, 2024-12-24:
@Pugwash1 

I have been working on updating the help pages for blocks and the new train part of Dispatcher.  With the addition of adhoc transits the help page no longer has a linear flow since it goes from train type -> transit type -> train selection -> transit details. 

I think something like the following might work better.

[image]

dsand47, 2024-12-24:
Here is another version.

[image]

The train data fields are located between the "Trains From" options and the "Transit Type" options.

Pugwash1, 2024-12-24:
@dsand47 
Sure, looks good.

dsand47, 2024-12-24:
> @dsand47 Sure, looks good.

If it is OK with you, I would like to cleanup the JPanel names and change the mainPane to a border layout.  This eliminates the automatic resizing of the header and footer button areas when the window size changes.

Pugwash1, 2024-12-25:
@dsand47 
I dont see why not, providing the central pane still scrolls as there are pretty small screens out there!
What would be good is to get rid of the top buttons and move them to menu items and/or an icon menu bar.

danielb987, 2025-02-03:
What's the status of this issue?

Pugwash1, 2025-02-04:
@danielb987  @dsand47 
Dave updated code around christmas, help pages close by. #13712

closed 2025-02-04 by Pugwash1