JMRI/JMRI #13717: Features to clear PanelPro LogixNG Clipboard & report presence of terms in Clipboard

JIM-MOOMAW, 2024-12-29:
I'm constantly using the clipboard to speed up editing, but it quickly accumulates hundreds of lines of text.  The only way I can see to clear it is to rt-click-Remove each line or block, which takes a *lot* of time and seems unnecessarily tedious.  Leaving all this code in it makes it hard to go find things, but is also an insidious tormentor when it contains sensors, turnouts and other objects not otherwise used you want to delete from the tables... you aren't thinking of things lurking there blocking a delete request.  

Request:  Is there a way to add a "clear clipboard" function to the clipboard Tools tab?  Please? 

Related Request:  The "where used" tool will report a sensor/turnout/etc term lingers "out there" in code-land but it doesn't say when things are hiding in the clipboard.  Is there any way the "where used" tool can mention when terms are in the clipboard so you can go purge them... ideally with the "clear clipboard" function... perhaps like it does with LogixNG references?   Once the terms are cleared out of the conditionals and still won't delete there's no further help where to look.

dsand47, 2024-12-29:
The "Tools -> LogixNG -> Where Used" menu item in the main PanelPro menu will show you which ConditionalNG has an item.  This includes the LogixNG clipboard.  

It appears that the checking for "in use" items such as sensors still occurs after a LogixNG expression or action has been deleted.  This is cleared after doing a Store followed by re-starting PanelPro and loading the xml data file.

Dave Sand

dsand47, 2024-12-29:
The "in use" problem appears to be a veto listener that has not been removed.

JIM-MOOMAW, 2025-02-17:
Any update on this request for a menu function to clear clipboard?   Up or down?  I can see there's a side conversation on the where-used function (my fault for including) but that wasn't my focus.  Tx.

closed 2025-06-30 by github-actions[bot]