JMRI/JMRI #13727: HM7000 Product IDs - wrong CVs used in class IdentifyDecoder

CdreKidd, 2025-01-01:
I am currently constructing an HM7000 TXS decoder definition file and there is a problem with the way JMRI identifies the Product ID - currently class IdentifyDecoder looks at CVs 47,47 and 49 - but these are the firmware definitions and relate to the chip, not the Sound Profile, which is needed to identify the CVs used for each locomotive/sound profile. The CVs that need to be accessed are CV 200 (High Byte) CV 201 (Low Bite) - in most instance the Low Byte (CV201) will be used as the product IDs are all less than 255. I think similar rules could be used as are applied to the TTS chips where CV 158 is an optionalCV.

It is important that the correct CVs are accessed for the Product ID as the Sound Definitions affect both the decoder functions and role of the CVs in controlling sound outputs.

Sadly I don't have any Java experience, and am unfamiliar with the working of GIT, so while I can (and have done) the file amendment, I can't work out what to do next!

bobjacobsen, 2025-01-02:
For discussion, see [this JMRIusers thread](https://groups.io/g/jmriusers/topic/110375807)

PUlvestad, 2025-01-02:
Related issue #12180

closed 2025-06-23 by github-actions[bot]