JMRI/JMRI #13731: LocoNet fast clock throttle ID is missing

fulda1, 2025-01-02:
Hi,

as per loconetpersonaledition.pdf very last paragraph:
> <ID1>,<1D2> This is device ID last setting the clock. <00><00> shows no set has happened
> <7F><7x> are reserved for PC access

JMRI does send IDs=00 when configure clock. That misleads devices, that recognize "clock is not set". It can be easily solved by adding one line to java/src/jmri/jmrix/loconet/LnClockControl.java line 341:
s.setThrottleIdentity(0x7F71);

Patch can look:

`*** LnClockControl.java.orig    Thu Jan  2 14:30:11 2025
--- LnClockControl.java Thu Jan  2 14:31:30 2025
***************
*** 339,342 ****
--- 339,343 ----
              s.setFcRate(curRate);
              s.setFcFracMins(curFractionalMinutes);
+             s.setThrottleIdentity(0x7F71);

              // set other content
`

Jindra

newHeiko, 2025-01-02:
Thanks for the report. I'll look into that sometime...

Is it the same throttle ID as used for every throttle operation or is this a special "clock device" ID? Because elsewhere we are using 02 71 as our throttle ID: https://github.com/JMRI/JMRI/blob/e62015a55ca7c2bce9bca525b7720fecc4e49677/java/src/jmri/jmrix/loconet/LnThrottleManager.java#L506 (0x171 will translate to ID2=02 ID1=71 on loconet, because those are not two bytes but two 7-bit values)

There seems to be a mismatch between OP_WR_SL_DATA / OP_RD_SL_DATA on page 10 of the loconet personal edition (which defines ID2=00 ID1=01 up to ID2=01 ID1=7f for PC usage, ID2=02 ID1=00 up to ID2=03 ID1=7f system reserved, ID2=04 ID1=00 up to ID2=7e ID1=7f for throttles and does not mention anything with ID2=7f) and the last page, which fixes ID1=7f. The last page makes no sense to me. Can anyone share any more information about this?

I could speculate they actually meant to fix ID2=7f for clock IDs, which would align with page 10 not mentioning ID2=7f at all, reserving it for clock slots?

Heiko

PS: Also I'd love if the ID wasn't hard coded and wasn't in the "reserved" part of the personal edition, but that's something else I've got on my todo list, and it's been there for years...

fulda1, 2025-01-03:
Hi Heiko,

In reality it does not matter, what ID is used. everything else 00:00 works fine.
My understanding is like this:
Once you have ID for throttle operations (stated on page 10 -> 02:71), you are welcome to use it for fast clock. Once you are running PC software, that control fast clock, without controlling locomotives (no throttle functions), it is recommended to use 7F:7x to avoid collision with other software throttles (last paragraph).

P.S.: Next ambiguous ID is defined in LNCV extension, this one is one byte only and JMRI uses 01.

Jindra

newHeiko, 2025-01-03:
Okay, so let's use the JMRI throttle ID for clocks too. I'll see that I can get that in soonish.

(The spec - in my understanding - suggests 7x:7f but as that does not make any sense, I agree they probably mean 7f:7x and your guess that's only for the computer if it's not controlling a locomotive sounds reasonable)

Heiko

P.S: If LNCV uses only a one byte ID, that's obviously a totally different thing. But clock commands are just "regular" slot writes to a specific slot, so those IDs should match...