JMRI/JMRI #13759: Externalise (and save) Turnout Circle Transparency

philguk, 2025-01-08:
When defining a turnout circle FILL colour, it is possible to also specify a transparency

But this only works for the CURRENT JMRI session. As the transparency is not saved in the panel xml file it does not persist across JMRI restarts

Could we please EITHER externalise the transparency setting OR (less desirable) remove the ability to set a transparency at all

Thanks

philguk, 2025-01-22:
I’ll leave this open for now, but it looks like this isn’t important enough to correct

Pity

dsand47, 2025-01-22:
This is a simple request can turn into a major project.

With the exception of Control Panel Editor shapes, colors are stored as 3 integers for red, green and blue.  The normal color selection process presents a list of standard Java colors with tabs for the extended Java color choosers which include transparency.

Adding the alpha (transparency) value to the layout data store process makes the xml file backward incompatible.  This can be quite disruptive.

Once we start changing this for turnout circles, are people going to want changes to all of the other places where color is used?

As a work around, I have developed a script than can be run after the loading the layout data file that can make a temporary change to the turnout colors.

[image]

```
import java
import java.awt.Color
import jmri

colorClosed = (0, 255, 0, 128)
colorThrown = (255, 0, 0, 128)

editorManager = jmri.InstanceManager.getDefault(jmri.jmrit.display.EditorManager)

for panel in editorManager.getAll(jmri.jmrit.display.layoutEditor.LayoutEditor):
    r, g, b, a = colorClosed
    panel.setTurnoutCircleColor(java.awt.Color(r, g, b, a))
    r, g, b, a = colorThrown
    panel.setTurnoutCircleThrownColor(java.awt.Color(r, g, b, a))
    panel.repaint()
```

philguk, 2025-01-23:
Thanks Dave

Is kind of assumed that adding something to the XML that prior versions weren’t expecting works just be ignored. Glad I didn’t try and make the change

When I get chance I’ll have a play with your script - handy that you used precisely the colours and transparent I was using!

Phil G

Screenshot.2025-01-22.at.4.06.46.PM.png (view on web)<https://github.com/user-attachments/assets/b24893a6-c017-4cfc-8036-44594f4a4a0c>

philguk, 2025-01-23:
And further thanks Dave

Script now installed and working just fine

Much appreciated

Phil G

closed 2025-03-12 by danielb987