JMRI/JMRI #13765: Document and expand the JSON WebSocket

ralflang, 2025-01-14:
As a developer

I want to write documentation on the JSON WebSocket and expose more of JMRI's state and event triggers
So that I can provide a solid basis for JMRI builtin or externally provided web apps and ultimately use the Java GUI much less.

Observing:

/web/ root folder contains some standalone examples of pages querying the WebSocket API as-is
/web/README.txt contains some general instructions
/web/servlet provides snippets to be processed/rendered through JMRI similar to PHP, cfengine & friends

The actual websocket server component is in package jmri.server.json 

TODO: Jot down the interaction between producing the schema definition for new exposed items, implementation of loading the items, client side calls...

icklesteve, 2025-01-14:
@ralflang this sounds great, I'm sure this will be of use.  If you require assistance just shout up :-)

nigelcliffe, 2025-01-21:
I'd support this.  

I've a few example webpages which use the JSON connections which I could contribute.   They're reasonably well commented, and not too complex, so might form the basis of examples.  

Please get in touch if wanting to discuss.

ralflang, 2025-01-28:
Status update from last weekend.

Initially I wanted something more serious but I settled for toying with the JMRI Railroad name because it is very concise.
First I setup a demo page in the vain of power.html but for the railroad name and then I demonstrated power.html does in fact react to changes in JMRI (or another browser windows) while the railroad name does not.

Then I went through the javascript client to identify the missing part for live updates of the railroad name through the socket.
Next I went through the client and server side ceremony for handling a **POST** railroad name update.

I also looked into how the web socket server interfaces with JMRI's internal event bus and took some notes on first-class endpoints (like turnout, power) vs second-class endpoints under the util package (like railroad name, ping/pong) and the meta schema server.

Once I have distilled instructions from this I will serve a first PR and then expand towards more interesting items of state.

I will reach out as soon as it makes sense and doesn't waste your time.

nigelcliffe, 2025-01-30:
For another use cases,  I've just written various webpages which might  fit into any tutorial/example pages, offering slightly more advanced use of calls into data held within JMRI.   They are:

- Use of RFID Tags to indicate location of stock on a webpage diagram (just RFID tags and Reporters, no other JMRI features). 
 
- Use of RFID Tags to indicate location of stock on a webpage diagram, plus the destination for the stock, the train the stock should travel on, and colour changes to indicate stock has reached correct destination.  (Requires RFID tags, Reporters, and use of Operations). 
 
- Table which is filterable to "owner" of stock, showing the last reported RFID location/time.  (Requires RFID tags, Reporters, and Operations (at least the "Cars" database for stock ownership)).     This example is fairly simple to follow the code.  

The above three have small images of the stock, but that requires someone to add all the PNG files to match the stock names.   

 
- WebPage which shows the Manifest view from Operations, allowing either individual "tick" of each task on the mainfest, or a "tick all", before marking the Train as "Terminate Train".      I'm tempted to put this in as a suggested new page for the main distribution, as it appears to do things additional to the normal webpage views of Operations.  

- Nigel

ralflang, 2025-01-31:
Hi Nigel, I am very much interested in this.

nigelcliffe, 2025-01-31:
Ralf,

attached to this are the final two examples which are simpler to pack as they're self-contained.    
Happy to discuss how they work, and how to explain them to users. 

[operations_examples.zip](https://github.com/user-attachments/files/18623356/operations_examples.zip)

closed 2025-11-03 by github-actions[bot]