JMRI/JMRI #13769: Removing a Node Hangs Further CDI Fetch For a Few Seconds

crosland, 2025-01-15:
The background is that I am using JMRI to test newly built nodes. Plugging them in one at a time for testing.

I plug a node into the network, fetch the CDI, do some testing and then unplug the node.

Plug the next node in. It identifies itself and appears in the network tree. I refresh the tree to remove the no longer connected node and close the configure window of the no longer connected node.

Open the configuration dialog by clicking on the new node in the network tree and the CDI fetch hangs until the second of the following messages appear in the system console:

> 16:28:59,318 enlcb.implementations.DatagramService ERROR - Overriding datagram transmit memo. old DatagramServiceTransmitMemo to 02.01.2C.01.34.00: 20.A8 new DatagramServiceTransmitMemo to 02.01.2C.01.32.00: 20.43.00.00.00.00.40 [AWT-EventQueue-0]
> 16:29:05,122 mplementations.DatagramMeteringBuffer INFO  - Never received reply for datagram 02.01.12.7F.44.A4 - 02.01.2C.01.34.00 Datagram: (2) 20.A8 [OpenLCB-datagram-timer]
> 

After which the CDI fetch proceeds and completes normally.

34.00 is the no longer connected node, 32.00 is the new node.

After which the CDI fetch proceeds and completes normally.

Is there some kind of monitor with a timeout that is looking for previously known nodes? If so it prevents new CDI activity if the node disappears.

This might not be an issue for a normally running layout in which case I am happy to live with it.

Andrew

bobjacobsen, 2025-01-15:
What seems to be happening is that closing the window sends a datagram to that node, but the node is not there.  Things then wait for a timeout on that datagram.

A workaround would be to close the CDI window before removing the 1st node.

I'll see if I can get the operations to work in parallel.  Might take a while, though.

crosland, 2025-01-15:
> From "Bob Jacobsen" ***@***.***>
> To "JMRI/JMRI" ***@***.***>
> Cc "Andrew Crosland" ***@***.***>; "Author"
> Date 15/01/2025 16:48:20
> Subject Re: [JMRI/JMRI] Removing a Node Hangs Further CDI Fetch For a
> Few Seconds (Issue #13769)

That workaround works - thanks.

bobjacobsen, 2025-01-16:
See Openlcb/Openlcb_Java#285