JMRI/JMRI #13772: Error in Reset values in Train-o-matic decoders

nigelcliffe, 2025-01-16:
Hopefully its quicker for someone to edit one line in each file than me to post them all here.  

There's an error in the "reset" statement inside most Train-o-matic decoder files.   A reset requires a value of CV8=8 to be effective.  
Train-o-matic manual says otherwise, but testing on a decoder shows that writing CV8=78 does not work, whereas writing CV8=8 does work.     ( I suspect an earlier author of the decoder file copied the CV8 default line value, with manufacture code of 78 ).  

TrainOMatic_FD.xml   Line 1546
TrainOMatic_Lokommander_2.xml    Line 102
TrainOMatic_Lokommander_2_p22.xml   Line 80
TrainOMatic_Lokommander_2_p22_v6.xml    Line 84
TrainOMatic_Lokommander_2_v6.xml   Line 104
TrainOMatic_NMJ_224.xml     Line 1071

Old line reads:  
<factReset CV="8" default="78">  

Replacement line should read:  
<factReset CV="8" default="8">

lbijlsma, 2025-01-17:
I’ll take care of this. 
The error was created by me and the repeat is because of copy and paste :-)

lbijlsma, 2025-01-17:
Can this be assigned to me - I don't have the ability to assign it to me directly.

nigelcliffe, 2025-01-17:
Thanks for fixing the error.

closed 2025-01-21 by danielb987