JMRI/JMRI #13785: Starting Help results in a loop;body=<!-- drag and drop the files in the open window onto here -->Issue Type: __Bug__Using PanelPro in JMRI version 5.8+Rbc21ce2ce7 with Jave version 17.0.13.When I start the "Help / Window Help" or the "Help / General

GerardWassink, 2025-01-18:
[profile.zip](https://github.com/user-attachments/files/18464658/profile.zip)
[session.log](https://github.com/user-attachments/files/18464659/session.log)
[systemInfo.txt](https://github.com/user-attachments/files/18464660/systemInfo.txt)

dsand47, 2025-01-18:
The main error is "java.io.IOException: Failed to bind to 0.0.0.0/0.0.0.0:12080".

On Ubuntu 24.04 running JMRI 5.10, the web server works as expected:
```
2025-01-18T09:48:34,142 server.WebServer                      INFO  - Starting Web Server on port 12080 [WebServer]
2025-01-18T09:48:34,215 server.WebServer                      INFO  - Starting ZeroConfService _http._tcp.local for Web Server with properties {path=/, json=5.4.0} [WebServer]
```

An unrelated problem is that you are running DecoderPro but trying to load a PanelPro layout data file.

I recommend that you try JMRI 5.10 and see if you get the same bind error.  If you do, you may need to determine if ZeroConf is installed.

dsand47, 2025-01-18:
You can also go to "Preferences -> Help" and select "Open help pages locally".  This uses the help pages included in the JMRI install.

closed 2025-05-05 by github-actions[bot]