JMRI/JMRI #13792: Signal Head size could not be adjusted down to 150%

EchoTango6, 2025-01-20:
I was placing a signal head on a Panel Pro Main Panel. I wanted to size at 150% but typed 250 by accident. When I tried to reset at 150% the signal head just got bigger and bigger. Had to delete signal head and place a new one on the panel.

Using latest production release.

EchoTango6, 2025-01-20:
> This is expected behavior.
>
> You first increased it by 250% to 250%, then again by another 150% to 1.5*2.5 = 375%
>
> If you want to make something smaller, you scale it by less than 100%. To undo a 250% scaling, you use 40% for example.

Ok, thanks for the clarification. I thought I was specifying an absolute percentage where the starting point default size is 100%.

I use this command with the command to set orientation. At first, I thought I was adding or subtracting degrees to the orientation but no, it does want an absolute number; in this a case bearing from 0-360 (where 0 defaults sideways from some reason).  I guess I can rotate the panel.

So, to me, the behaviors seemed a bit inconsistent.

Eric G Troup
Email: ***@***.***
Cell: +1 678 699 2348

dsand47, 2025-01-20:
Eric,

The default signal head images are from the "left" group.  

Dave Sand

EchoTango6, 2025-03-10:
Close. The issue is resolved.

closed 2025-03-10 by EchoTango6