JMRI/JMRI #13811: Web server does not recognize shape POLYGON

StCircumstance, 2025-01-24:
In Control Panel Editor, the polygon tool is used to draw a simple straight line however it is not rerecognized on the web server. Also, three seems to be an issue with the draw order or layer of the ellipse shape. See attached xml excerpts and photos of panels as viewed from the JMRI computer and the web server.

[image]
[image]
[PolygonEllipseXML.txt](https://github.com/user-attachments/files/18539707/PolygonEllipseXML.txt)
[image]
[image]

mstevetodd, 2025-01-24:
@StCircumstance Please attach a simple xml file that can be used to reproduce the issues described.

StCircumstance, 2025-01-24:
Hi Steve. I attached a TXT file with both the polygon and ellipse snips
from my XML. Is that not sufficient?

mstevetodd, 2025-01-24:
An intact XML file will allow myself (or others) to simply load it and see the issue, without having to guess at and create the missing portions of the file.

StCircumstance, 2025-01-24:
[NY&HV.txt](https://github.com/user-attachments/files/18543860/NY.HV.txt)

attached XML renamed to TXT

For the Polygon (line) issue go to Controls... Scheme
For the Ellipse issue go to About... Region2

Thank you Steve

mstevetodd, 2025-01-25:
Thanks, Vinny.
The Region2 panel console log shows "unhandled widget type of 'positionablePolygon'" so it's clear that new code will be needed to draw positionablePolygons.
I could not duplicate the problem with the ellipse, I assume because I do not have your custom icons on my machine. The issue is with the level of the drawn ellipse vs. the icon, correct?

It would save time if your example panel was a simplistic one, just showing these issues, and included any custom icons required.

--SteveT

StCircumstance, 2025-01-26:
Hi Steve, thank you for looking into these issues. Yes, the ellipse issue I believe is related to the levels. I forgot you'd need my icon to make it work. Even if the elliipse is drawn after the icon and the level of the ellipse is higher than the icon, the icon remains on top. Should I just send my icon or should I create a new panel with only the icon and the ellipse?

StCircumstance, 2025-01-26:
[image]
[NY&HV.txt](https://github.com/user-attachments/files/18551013/NY.HV.txt)

Hi Steve, attached icon and xml.
I created a new panel AAA_Test.
On the panel I added the icon first, then an ellipse and then a circle.
The default level for all objects was 3.
I then changed the level of the 2 shapes to 5.
On the JMRI computer the shapes are on top of the icon but on the web server the icon remains on top of both shapes.

bobjacobsen, 2025-01-27:
See also #12794 Web panel ignoring element "level"

closed 2026-02-02 by github-actions[bot]