JMRI/JMRI #13814: Decoder file for Rampino loco-1

DieterFlunkert, 2025-01-26:
This is the decoder file for a Rampino Elektronik Loco-1.

[RampinoElektronic_Loco-1.xml.zip](https://github.com/user-attachments/files/18551393/RampinoElektronic_Loco-1.xml.zip)

lbijlsma, 2025-01-28:
Hi Dieter,

As I can't find any emails on the JMRI mailing list about this. Do you want this to be added to the JMRI decoder definitions?
I can take care of this. Just want to make sure that is what is expected. I also don't have a test decoder so can't test whether it works but can definitely make sure that both decoder definitions are added to JMRI.

Let me know.

Lolke

DieterFlunkert, 2025-01-28:
Hi Lolke,Yes, I would like that the decoder definitions are added to JMRI. I testet the loco-2 one. The other one should be nearly the same except minor ***@***.***@icloud.com

lbijlsma, 2025-01-28:
Let me have a look at it. I found the manual and checked the loco-1 definition. It needs quite a few changes. I will try to address them and get this uploaded later this week.

closed 2025-02-06 by DieterFlunkert