JMRI/JMRI #13826: Failure to find locos in some blocks;body=<!-- drag and drop the files in the open window onto here -->Issue Type: __Bug__I am using a Lenz LZV200 with 4 LRB08 block occupancy modules. These are connected to 27 blocks on my layout. When I start JMRI an

FlexibleDave, 2025-01-29:
[profile.zip](https://github.com/user-attachments/files/18591102/profile.zip)
[session.log](https://github.com/user-attachments/files/18591100/session.log)
[systemInfo.txt](https://github.com/user-attachments/files/18591101/systemInfo.txt)

bobjacobsen, 2025-01-29:
The first part of the @FlexibleDave's description was in the title field.  See below.  Unfortunately, it was truncated.

> Issue Type: Bug
> I am using a Lenz LZV200 with 4 LRB08 block occupancy modules. These are connected to 27 blocks on my layout. When I start JMRI an

FlexibleDave, 2025-01-29:
I am using a Lenz LZV200 with 4 LRB08 block occupancy modules. These are connected to 27 blocks on my layout. When I start JMRI and load  the layout only the locos in blocks connected to the first LRB08 (65:1 to 65:8) are registered. I notice in the system log the UM1 (sensor 68:7) is marked at empty when it is definitely occupied. 
If I start JMRI and load the layout and then switch on the Lenz controller there is no problem and all the locos are found  in the correct blocks.

closed 2025-04-21 by github-actions[bot]