JMRI/JMRI #13828: Updating JSON-SCHEMA from Draft-4 to Draft-7 or higher

ralflang, 2025-01-30:
# Proposal 

JMRI's JSON APIs for HTTP and WebSocket protocols are currently exposed as schema definitions in the JSON-SCHEMA standard draft-4
The oldest JSON-SCHEMA standard currently still advertised on the json-schema.org website's specification widget is draft-5, published in October 2016.

[image]

I would like to update JMRI's JSON Schema metadata to a newer draft. 
Depending on feedback to this issue, I would target 

- draft-7, published in March 2018,  https://json-schema.org/draft-07/draft-handrews-json-schema-01
- or JSON Schema Core, published in December 2020 https://json-schema.org/draft/2020-12/json-schema-core

# Motivation

JSON Schema draft-4 has been superseded by multiple later drafts in a mostly backwards compatible way. It's no longer a JSON Schema version a developer would target when building new API versions. My key observation triggering this is trivial: I wanted to explicitly mark a message as "all data fields are optional, none are required". This is possible from draft-6 onward by providing the optional required attribute with an empty list value. In draft-4, json-schema would not allow the "required: []" notation. 

# Impact / Risk

From reviewing JMRI's use of json-schema, the upgrade should be straight forward and neither break end internal consumers (i.e. the JMRI UI) nor external consumers (third party web apps). Both proposed target versions are old enough to be supported in mature versions of typically used consumer libraries. It only affects the metadata service, not the API format itself

# How to validate

- Formal schema validation (already in the CI pipeline)
- End user test with the Web UI.
- Some interactive API integration tests (to be automated/scripted for future CI usage?)

mstevetodd, 2025-01-31:
Not an area I have any expertise in, but @ralflang's proposal seems very thorough and well-thought-out.

danielb987, 2025-01-31:
I agree.

rhwood, 2025-01-31:
There is no reason not to adopt newer versions of the schema.

danielb987, 2025-04-28:
I have reopened the PR since I think it would be good if @ralflang would like to update the schema.

ralflang, 2025-04-30:
Todo.

closed 2025-07-28 by github-actions[bot]