JMRI/JMRI #13850: Yamorc support

HenkEdel65, 2025-02-06:
Issue Type: __Feature Request__

When will JMRI support the new YAMORC equipment, f.e. the YD7010 central station.

DecoderPro 5.10+Rca461bd266
OpenJDK 64-Bit Server VM BellSoft 11.0.26+9-LTS mixed mode, sharing
Windows 10 10.0 amd64

HenkEdel65, 2025-02-11:
> For the central station, have you tried the options that are documented on the Digikeijs DR5000 page?
> https://www.jmri.org/help/en/html/hardware/digirails/index.shtml
>
> It would be my assumption that all these options would work for Yamorc as well. There is no dedicated DR5000 support and all the connection options that the DR5000 supported are also supported by the YD7010.
>
> Lolke

Hi Lolke,

 

Thanks for your answer. I tried your suggestion, but it doesn’t succeed till now.

 

1.	The first thing that strikes me is the message Unknow Track Voltage:

 

 

 

 

2.	I used the setting from the helpfile:

Via USB:

System Manufacturer: Digitrax
System connection: LocoNet LocoBuffer-USB
Serial Port: (as needed for your USB connection)
Command station type DSC100 (Chief)

 

The selection of COM4 results in this error:

 

 

 

The only serial port is COM1, that starts the program JMRI apparently without errors.

But this makes no sense and then still it doesn’t work.

 

3.	The system console says:

 

log4j:WARN No appenders could be found for logger (apps.util.Log4JUtil).

log4j:WARN Please initialize the log4j system properly.

log4j:WARN See  <http://logging.apache.org/log4j/1.2/faq.html#noconfig> http://logging.apache.org/log4j/1.2/faq.html#noconfig for more info.

 

And this is too technical for me….

 

What can I do?

 

Kind regards,

 

Henk Edel

silverailscolo, 2025-02-11:
As you mention not all COM ports are recognized, did you check the VCP drivers are working? See the [link](https://ftdichip.com/drivers/vcp-drivers/) on https://www.jmri.org/help/en/html/hardware/digirails/index.shtml

HenkEdel65, 2025-02-12:
Thanks for your answer.I need Com4 and that works from the Yamorc 7010 Command Station.But it does not with JMRI.Do I need to replace this driver?Kind regards,Henk Edel

silverailscolo, 2025-02-12:
Henk, when you describe
“COM4 works from the Yamorc 7010 Command Station”
do you mean you use a Windows program to configure the Y7010?

In that case I suggest you close that program to release the (virtual) com port before starting JMRI. Perhaps even restart, depending how Windows handles this (I am on Mac).

HenkEdel65, 2025-02-12:
In a later reply I wrote that I got it woking with the new drivers and Com5.You can close this. Issue.Thanks!Henk EdelVerstuurd vanaf mijn iPad

lbijlsma, 2025-02-13:
Great that you got it working!

closed 2025-02-13 by silverailscolo