JMRI/JMRI #13851: Feature add to the anyrail script AnyRailExportAdditions.py #anyrail #layout editor

DeepRunRR, 2025-02-06:
I have really enjoyed being able to import my anyrail layout changes to JMRI. I use the AnyRailExportAdditions,py script to add all the sensor. block and turnout names and designations simply and accurately. When I import the anyrail file all of the track segments are set to mainline on the layout editor. Although this is an easy change to remove mainline from non mainline track it does involve a lot of manual updates.  
 
Now for the requested feature. In the anyrail section creation window there is a field called description that in anyrail version 7.6.4 can be manually defined. I think in previous versions they were set values. The default description is Unspecified. 

What I would like to know is if the text in this field could be used by the AnyRailExportAdditions,py script to set the mainline flag in JMRI for a section of track. For my case if "mainline" is in the box then the mainline definition is used in layout editor for the track segment.  

Here is the xml output by the anyrail export program. The word mainline in the blockname is the data I entered in the anyrail section field.

blockname="B_107_1024 TK 3 CSX RFP Bridge_**mainline**" dashed="no" mainline="yes" hidden="no" class="jmri.jmrit.display.layoutEditor.configurexml.TrackSegmentXml"></tracksegment><tracksegment ident="F1166-S-0" connect1name="A470" type1="POS_POINT" connect2name="A474" type2="POS_POINT" 
 
Thanks for your consideration
 
Mike

JMRI groupsio link to more discussion. https://groups.io/g/jmriusers/message/239474

DeepRunRR, 2025-02-06:
Hi Bob,

I do not think there is a spot in anyrail where the user has access to change the mainline flag field. I was hoping the anyrail section field would do that but it just appends the text to the block name. Steve Todd and I have been discussing this in the groupsio forum and he suggested an issue be generated. 

I would be guessing what anyrail uses to set the values for the JMRI settings of dashed mainline and hidden.

Mike

DeepRunRR, 2025-02-06:
Bob 

I posted the selection question on the anyrail forum to see if there is  a setting for this.

Mike

DeepRunRR, 2025-02-13:
I received a response from the Any rail post and there is not a setting or anything else in any rail related to setting mainline.

 Mike

mstevetodd, 2025-05-13:
It appears that the Description is now optional as well, which breaks my script. So, while fixing that, I'm looking at this request as well.

@DeepRunRR 
One problem is that the current AnyRail export defaults to `mainline="yes"`, and I do NOT want to change the default by requiring "mainline" in description. So, what is the opposite of "mainline"? "branch", "spur", "siding" all seem to have additional considerations. "notmainline" seems correct but terrible. "secondary"? I'm going to go with "secondary" until I hear a better name.

KenC57, 2025-05-13:
If you follow most prototype books, "other track" would be the
opposite of mainline. Sidings (as used for passing by the dispatcher)
are a gray area between the two.

 

-Ken Cameron, Member JMRI Dev Team

www.jmri.org <http://www.jmri.org/> 

www.fingerlakeslivesteamers.org
<http://www.fingerlakeslivesteamers.org/> 

www.cnymod.com <http://www.cnymod.com/>

DeepRunRR, 2025-05-21:
Todd recall all I wanted was to not have the import set all the track in the JMRI panel to mainline.
Todd I also think I am finished importing anyrail into JMRI as I have completed my signal work and I don't think that will transfer to a new import.
Thanks Mike

clockbrain, 2026-10-25:
I also found this AnyRail export defaulting of all sections to "mainline" a bit of a chore to fix. The real solution is obviously for AnyRail to fix their XML export but in the meantime here is a simple patch to AnyRailExportAdditions.py that flags all descriptions ending in "_other" as not mainline. It ignores descriptions ending in "_mainline" and puts any other trailing value into the block description as per original behaviour.
Replaces the code starting around line 85.

`    anyRailDescription = None
    if (len(ra) == 4) : # user defined, "Unspecified", "mainline", "secondary", etc.

        if (ra[3] in ["mainline","other"]) :
            if (ra[3] == "other"):
                # Find all track segments belonging to this block
                for trackSegment in layoutEditor.getTrackSegments() :
                    if (trackSegment.getBlockName() == block.getUserName()) :
                        trackSegment.setMainline(False)

                log.debug('Block "{}" set as non-mainline'.format(blockSystemName))
        else:
            anyRailDescription = ra[3]`