JMRI/JMRI #13868: JMRI 5.10 LogixNG almost-newby how to use references

Tom-Douglas, 2025-02-14:
Hi, Folks!

I've been dabbling with LogixNG for a few months, mainly shutting off track power when the trains get on the wrong. track.

I have a project in mind that will ID the train, throttle it down if needed, then sense two blocks ahead, checking for a second train ahead of it.  This is to maintain a good following distance.  I have a 2-D array loaded in to Logix NG Array, named Block_Clearance.

What I'm wrestling with is getting references and double references into either local variables or memory.

Line 7 shows that I can copy Block_Clearance[i,Memory (i being index, Memory being the memory that I want to load to) into local m.  Successful per debug.

Line 8 I load local m into memory Test #1.  Successful, however I want to double reference "{{m}}" into Test #1 which errors out.  I need a way to double reference m mbut I see ther's no reference for the destinator.  What an I doing wrong?

Lines 9 and 10 same story.  I need a double reference "{{LSx}}"; they error out.

Line 11 is a local variable readout.

Line 12 returns False regardless of the sensor status.  Local i contains the sensor designator.  How would I get the condition of the sensor?

Thank you for your help!
Tom D.
Sabetha, Kansas

-----------------------------------------------------------
```
1    ConditionalNG: Test #12
Local variables
#i	String	0	Select
s	String		Select
sb1	String		Select
sb2	String		Select
m	String	0	Select
m2	String	0	Select2        ! A
            If Then Else. Execute on change
3                ? If
                    Sensor Test Sensor #1 is Active
                ! Then
4                    Many
                         ! A3
5                            Table: For each row of column "Sensor" in table "Block_Clearance" set variable "i" and execute action A1
6                                ! A1
                                    Many
7                                        ! A3
                                            Set variable m to the value of table "Block_Clearance" row "by local variable "i"" and column "Memory"
8                                        ! A8
                                            Set memory Test 1 to the value of variable m
9                                        ! A5
                                            Set variable sb1 to the value of table "Block_Clearance" row "by local variable "i"" and column "Ahead_1"
10                                        ! A6
                                            Set variable sb2 to the value of table "Block_Clearance" row "by local variable "i"" and column "Ahead_2"
11                                        ! A4
                                            Log local variables
                                        ! A1
12                                            If Then Else. Execute on change
                                                ? If
                                                    Conditional by reference "{i}" is True
                                                ! Then
                                                    Many
13                                                        ! A1
                                                            Set memory by memory "Test 1" (No Listen) to the value of table "Block_Clearance" row "by local variable "i"" and column "Block_Name"
                                                ! Else
                                                    Many
                                                        ! A1
                                                            
                                        ! A2
                                            
                        ! A1
                            
                ! Else
                    Many
                        ! A1
                            

```----------------------------------------------------
```
Block_Clearance
Sensor	Reporter	Memory	Block_Name	Ahead_1	Ahead_2	Seq
LS1	LR1	IM1	West Turn	LS3	LS13	1
LS3	LR2	IM2	Westdale	LS13	LS17	2
LS5	LR3	IM3	Heartland 2	LS11	LS21	3
LS7	LR4	IM4	Heartland 1	LS19	LS21	4
LS9	LR5	IM5	Southgate Main	LS1	LS3	5
LS11	LR6	IM6	Southgate Siding	LS1	LS3	6
LS13	LR7	IM7	Northport A	LS17	LS19	7
LS15	LR8	IM8	East Park Spur 2	LS19	ls23	8
LS17	LR9	IM9	Northport B	LS19	LS23	9
LS19	LR10	IM10	Eastman	LS23	LS9	10
LS21	LR11	IM11	East Park Main	LS19	LS19	11
LS23	LR12	IM12	Eastbach	LS9	LS1	12

```

dsand47, 2025-02-14:
It would be a lot easier to comment on your issues if you upload the layout data xml file (use a .txt suffix) and the CSV file.

danielb987, 2025-02-15:
I think you should use a `Digital Formula` action instead of a reference. In the formula, you can use the function `readMemory(memory)` to read the value of the memory.

Tom-Douglas, 2025-02-17:
Hmmm....  I don't know what I'm doing...  Could you folks give me an idea how to upload the XML and CSV files?  I'm trying a few different links and pages, just not finding anything.

Well, I took a shot and loaded Douglasville.xml and Block_Clearance into @Tom-Douglas ➜ /workspaces/codespaces-blank

Thanks!

dsand47, 2025-02-17:
The upload is at the bottom of the comment.

[image]

Tom-Douglas, 2025-02-17:
Thank you, Dave!  That was simple enough! :)
Code is below.
Should I remove my uploads in the codespaces?

Tom D.

[TomD Test Douglasville.xml.txt](https://github.com/user-attachments/files/18831345/TomD.Test.Douglasville.xml.txt)

[Block_Clearance.csv.txt](https://github.com/user-attachments/files/18831358/Block_Clearance.csv.txt)

Tom-Douglas, 2025-02-17:
> I think you should use a `Digital Formula` action instead of a reference. In the formula, you can use the function `readMemory(memory)` to read the value of the memory.

Daniel, the readMemory(memory) function makes sense to use.  Would I use the Digital Formula for acquiring a sensor status too?  How would I do that reading a value from the Block_Clearance table?

Thank you for the help!

Tom D.

dsand47, 2025-02-17:
Tom,

I don't know what the goal is but there are a couple of things that might make a difference.

Here is a piece that I re-worked.

>                                        ! A2
>                                             If Then Else. Always execute
>                                                 ? If
>                                                     Sensor by local variable "i" is Active ::: No Listen
>                                                 ! Then
>                                                     Set memory by local variable "m" to the value of table "Block_Clearance" row "by local variable "i"" and column "Block_Name"
>                                                 ! Else
>                                                     Set memory by local variable "m" to ""

I recommend that you do not try to mix Logix and LogixNG, such as using the Conditional expression.  They work differently.

Since you are doing a "For-each" loop, any "If-Then-Else" within the loop need to use "Always Execute" and the expressions need to use "No Listen".  

The Sensor and Memory references use the "Local Variable" tab instead of the "Direct" tab.

An alternate approach uses the table for the memory destination and the block name.

Some of these ideas might make no sense based on the real context, but they might be helpful.

Dave Sand

danielb987, 2025-02-17:
@dsand47 
Thanks for your reply. But the formatting of your answer got really bad so it would be good if you could write your answer directly on GitHub instead of reply by email.

dsand47, 2025-02-17:
I don't know what the goal is but there are a couple of things that might make a difference.

Here is a piece that I re-worked.

```
! A2
    If Then Else. Always execute
        ? If
            Sensor by local variable "i" is Active ::: No Listen
        ! Then
            Set memory by local variable "m" to the value of table "Block_Clearance" row "by local variable "i"" and column "Block_Name"
        ! Else
            Set memory by local variable "m" to ""
```

I recommend that you do not try to mix Logix and LogixNG, such as using the Conditional expression.  They work differently.

Since you are doing a "For-each" loop, any "If-Then-Else" within the loop need to use "Always Execute" and the expressions need to use "No Listen".  

The Sensor and Memory references use the "Local Variable" tab instead of the "Direct" tab.

An alternate approach uses the table for the memory destination and the block name.

Some of these ideas might make no sense based on the real context, but they might be helpful.

Tom-Douglas, 2025-02-19:
Guys, thank you so much for your help!  I'll try and clarify things a little.

The LogixNG of interest is found in LogixNG Test, Test #12.  I added comments in this script to help clarify my intentions.  The script is far from finished; I'm only seeking the key actions and expressions that I'm not familiar with.

The objective of this event is to:
1, locate a target locomotive, 
2, sense two blocks ahead for a train ahead of it, and slowing the target train if a train is ahead (following the train ahead),
3, sense one block ahead for a train, and stopping the target train so it doesn't collide with the train ahead
4, display the block and speed of the target train.
NOTE: The event would be triggered by the reporter when it changes.
This event is "Test 1" sensor triggered for debugging purposes. The toggle button is found on the String Test Panel for convenience.
I'm just trying to get key actions figured out so I can set this event up decently.

The first question is this: Is there a reference in JMRI that shows examples of what I'm seeking?  I have a feeling that I overlooked those.  For example, I looked at LogixNG tables in JMRI's LogixNG Reference - Chapter 11, LogixNG Tables.  It shows how to use a table but so far I haven't been able to apply things like double references successfully.  That's where I'm hung up.

Finally, the csv and xml files are found below.

Thanks again for your help!
Tom D.

[TomD Test Douglasville.xml.txt](https://github.com/user-attachments/files/18862748/TomD.Test.Douglasville.xml.txt)

[Block_Clearance.csv.txt](https://github.com/user-attachments/files/18862750/Block_Clearance.csv.txt)

dsand47, 2025-02-21:
I assume your double reference is an attempt to copy a memory value from a memory name from a table to a specific memory name.  Try this:

[image]

Tom-Douglas, 2025-02-22:
#I assume your double reference is an attempt to copy a memory value from a memory name from a table to a specific memory name.

[image]

OK, this works, it's exactly what I need.

The only thing left is testing the sensor status at A1.  I changed that to what was suggested.

[image]

The result comes back as "False" regardless of the sensor status.  I'm not sure what "Listen/No listen" does -- I should find that in the documentation.

We're getting close! :)  Thanks again!

Tom D.

dsand47, 2025-02-22:
You are trying to call a Logix Conditional from LogixNG.  While this is supported if done correctly, it rarely has any value over doing  actions directly in LogixNG.

This returns "false" since there is no Logix Conditional with a name that matches the sensor name.

[image]

This works much better.

[image]

Tom-Douglas, 2025-02-22:
That one works!  Thanks so much, Dave!  I'm off and running!

It's hard to tell what is Logix style and the LogixNG.  More reading in the documentation.

Thanks again!

Tom D.

closed 2025-02-22 by Tom-Douglas