JMRI/JMRI #13870: NCE Consist Backup/Restore exceeding the memory range

KenC57, 2025-02-16:
It looks like since 5.7.1, when I made the changes for the new memory mapping in the PH5, it messed up the memory range used for the consist backup. Problem is it made the backup file about 0x200 more than it should. Problem on the restore is that didn't test to see the value out of range and would overwrite things like the AIU and RS-232 buffer memory areas.

I think I've got it fixed for the backup and a patch to stop the restore at the right place regardless of the contents in the file. I hope to have more sometime tomorrow.

But the issue is all versions since 5.7.1 have the problem. Since it has taken over a year for it to surface, I'm rating it kind of low in general. But I'm thinking once I have the fix, I'll post on the NCE site for those active users to work with the fixed build and how not to use the other versions for consist backups.

Which release notes should I stick a notice into once I've got the fix?

KenC57, 2025-02-16:
Just to make sure, I went back and tested with 5.7.1 and 5.8 to confirm the problem goes back that far.

Now it looks like I can go into the JMRI/website tree and do an edit of the different release notes. Or should I really have git pull that tree and then edit, push, and then make a pull? Seems a bit much for just adding the warning line to those pages.

bobjacobsen, 2025-02-16:
@KenC57 Whichever you prefer.  I find editing locally and then creating a PR to be simpler. That way, I can use my favorite editor and copy/paste the warning into the five files.

KenC57, 2025-02-16:
Entered pull  #13871 to deal with this. Also updated 5.7.1, 5.8 and 5.10 release notes pointing them to 5.11.4.

closed 2025-03-01 by KenC57